betaflight/betaflight-configurator #2882: Need builds for Apple Silicon (arm) chips.

crewshin, 2022-04-06:
### Is your feature request related to a problem? Please describe

No "problem" other than becoming outdated fairly quickly. These machines are incredible and this is the path forward for Apple Hardware.

### Describe the solution you'd like

We need a compiled binary for Apple's new M1 chips to run natively.

### Describe alternatives you've considered

NA

calif94577, 2022-04-24:
While not absolutely necessary as it can run on Rosetta2 I concur that this would be a nice to have as an M1 Pro chip owner.

crewshin, 2022-05-02:
"While not absolutely necessary"... yet. Rosetta has always been a temporary transition during architecture switches. I agree with you at the moment, however, this is our chance to prepare for the cut.

haslinghuis, 2022-06-02:
No idea if this is even runs on a M1 so please check

haslinghuis, 2022-06-03:
@crewshin using the PR (but unable to test myself) you could test it yourself if you setup a dev environment

crewshin, 2022-06-03:
Sorry about that. I misunderstood the context as I was on my phone. Will try and give it a shot soon.

gandlers, 2022-06-20:
I have tried to build for ARMv8 on ManjaroARM.
Tried to cherry pick this commit into the 10.8.0 release of configurator from the main betafight git hub but not a git user so having limited success.
There are plenty or warnings for storybook addons and I have noticed that the new nwjs builds for arm have very different contents between v0.27.6' and the new'nw60-arm64_2022-01-08 version but I dont know enough about java, NWJS yarn gulp and any knowledge to work out how to fix it.
when I have run yarn gulp release --armv8 the elf file created looks like standard linux32 not arm64.

L1cardo, 2022-06-21:
"Rosetta has always been a temporary transition option during architecture switches which will be removed at some point" is a thing, but what matters more is the performance.

closed 2022-07-05 by blckmn