betaflight/betaflight-configurator #2885: Missing items in config 10.8 if using targets under 4.3

dutchschaefer, 2022-04-08:
### Describe the bug

Options missing or not able to save or change certain things in 10.8 RC4 when using a quad with 4.2.9.  

### To Reproduce

Trying to setup bi-directional Dshot and the options will not save.  Dmin in PIDs tab is also missing.

### Expected behavior

See reproduce section.

### Configurator version

10.8 RC 4 or any other than 10.7

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-04-08:
Testing with 4.2.11 now.

D min is now called Dynamic Damping.

[image]

Also enabling Bidirectional DShot does work for me:

[image]

Please report the diff being used.

dutchschaefer, 2022-04-09:
Ok, I see the dynamic dampening now, but I still was not able to turn on bidirectional dshot in 10.8 RC3 when using a 4.2.9 config.  It would allow me to select the switch but after rebooting BF, the option would be back to its original turned-off setting.  I had found someone online that mentioned using the 10.7 config to make the changes and that worked.  I tried again with 10.8 RC4 and got the same results.  Ended up moving back to 10.7.  I do wonder if it has something to do with the macbook air as I was only able to use 10.7 properly on an Imac.  Using it on the air and it would crash.  I even downloaded the version BF suggested(I think it was a nightly, its been a while) and that worked but I couldnt get some of the options to work.  Anyway, I am going to close this issue for now.  Hopefully the final release of 10.8 will fix any issues when using the MB air.

closed 2022-04-09 by dutchschaefer