betaflight/betaflight-configurator #2888: Persist Presets In Setup

MythicLionMan, 2022-04-09:
### Is your feature request related to a problem? Please describe

Once a set of presets have been applied to an FC the presets that were selected, the order in which they were applied, and the options that were selected are all 'lost'. If you want to change one of the options on a preset there is no easy way to do so without reconfiguring from scratch or reverse engineering the presets that were originally applied.

Migrating to a new version of Betaflight is also problematic, because some parameters are safe to copy from one version to the next, while others are not. Presets go a long way to addressing this, because they are updated along with the major Betaflight version, but there is no way to know which presets were used when the quad was initially setup.

### Describe the solution you'd like

I'm proposing making the presets that are applied to the FC a persistent part of the quad setup. When presets are applied the presets and their options are stored in a parameter on the FCs flash. When the presets tab is revisited the original set of presets and their configuration are loaded.

The preset configuration is stored in a new Betaflight parameter, `preset_configuration`. `preset_configuration` is a text parameter. It contains the path to each preset, followed by the the names of any options that were applied and a newline. For example:

```
filters/default_rpm_clean.txt 
tune/SupaflyFPV_Freestyle_6_and_EasyTune.txt 'RPM filtering' 'With HD Cam'
```

When the presets tab is loaded the parameter is read and the original presets are added back to the queue. The presets can be reconfigured, disabled, etc, and then re-applied.

There is a new option on the 'Flash Firmware' tab that reads `preset_configuration` before erasing, and sets it back after installing the new firmware. The setup can be restored by going to the presets tab and re-applying the appropriate version of the original presets onto the new version of Betaflight. If there is no up to date version of the preset for the new version then the most recent version would be selected instead.

A stretch goal would be to have a button to save the current quad configuration as a new 'root' preset that can be committed to a fork of the presets repository and used to configure other similar quads. The full quad setup is now defined by `preset_configuration` plus any parameter values that are different from the final state of applying those presets. The new prset would start with includes to recreate the original configuration stored in `preset_configuration` and then restore the custom parameters as explicit property updates. By saving a preset that replicates the current setup it's possible to easily make a 'personal setup' preset that will configure all of your quads the same way.

### Describe alternatives you've considered

If there was a way to identify an individual FC then `preset_configuration` could be stored locally and indexed with the unique ID. This could be a serial number (if the FC has one) or a new property with a GUID for the FC that could link the saved preset state with the original FC. The downside to this distributed approach is that the FC always has to be configured from the device where the configuration is saved in order to update the presets. This could be a fallback mode for low memory devices that cannot store the full preset configuration.

Another way to get a similar functionality would be to save the preset configuration locally on the host computer instead of on the FC. There could be a library of 'preset packs' stored this way. But that ends up being a re-invention of the preset system, because the 'preset pack' is essentially another preset. It also looses one of the chief benefits of this system which is to allow the original configuration of the FC to be adjusted without having to start from scratch.

### Other information

While rather simple in application, there is a paradigm shift in this change. Presets go from being 'actions' that you perform on an FC once to being a persistent base for the configuration of the FC. This is compelling, but I'm not sure if this a direction that the project wants to shift. There are currently 'tool' presets that do things like enabling or disabling GPS on a quad that doesn't have it. In this new system these tools would have to become part of the stored configuration, or there would need to be a way to specify that a preset was being applied as a one off and should not be persisted.

There are a few other issues with this proposal.
* The `preset_configuration` parameter can get quite large if many presets are applied. I'm pretty sure that this would quickly run into storage limits on low end FCs. There are various ways to compress the paths (for example, all of the path prefixes are known in advance, so they could be tokenized. Another optimization is that the preset options are very verbose at the moment and would benefit from having shorter representations for this feature and others). But even with multiple compression methods this would still most likely be the largest stored parameter in Betaflight, and might not be possible to store on the available hardware.
* There will always be some values configured that are not a part of the presets (including adjustments to the preset values). These values will be lost when re-applying the preset stack. If the presets are re-applied without flashing it would be possible (but not trivial) to find these 'overlay' values by diffing the current state with the original preset state and then applying them back. In a situation where the FC is erased and the original presets are restored the overlay values would be lost.
* Serial ports need to be configured before presets are applied, or certain settings are ignored. For instance, if there is no port defined for GPS and GPS is enabled, the setting doesn't stick. This means that the serial port configuration needs to be applied before the presets or the presets are not applied correctly. It's likely that there are other parameters that have the same requirements.

haslinghuis, 2022-05-10:
ping @limonspb

atshort, 2022-05-16:
Would it also be useful as part of this proposal to record the preset version used? I think if a preset is updated after first release this can result in two quads having a different configuration if configured at different times with the same presets.

MythicLionMan, 2022-06-06:
> Would it also be useful as part of this proposal to record the preset version used? I think if a preset is updated after first release this can result in two quads having a different configuration if configured at different times with the same presets.

That's a good idea. Storing the version number would also allow us to diff the current version of the preset with the version that was originally applied. This would obviously be another tool to write, but it would be quite handy to see what has changed in the preset when reconfiguring.

TehllamaFPV, 2022-06-06:
To me, the biggest limitation of this would be volatility, particularly for 'local version' type management, where that data is actually being stored (especially when combining non-BF-Master repository preset sources... and most particularly relevant when testing something which might be part of an active PR).
I really like the idea of this, and for most standard use cases having what amounts to an event log of presets applied would be a pretty solid thing, especially since it needn't be that verbose in actuality.
The trickier part becomes what densified notation would make sense for options present (and how this would transfer from preset versions, i.e. it would probably need to either be pretty verbose and list full JRadio Name boxes of each 'CHECKED' option, or use some other scheme to track which options are selected, the latter reducing readability significantly.
In terms of intent, it's nice to be able to export and transfer a 'what Presets were applied in what order' type thing, which that could accomplish, but I see that as an interim solution compared to what you're describing as the configurator-wide paradigm shift possible... but for the latter, I'd caution primarily that the Preset system is an intensely elegant solution for preparing CLI snippets to just independently run through the configurator, and function nearly independently of other configurator functionality.  Having baked-in version management and other things would have to be built into the preset system, and likely end up increasing the required maintenance effort for incrementing overall Betaflight versions and such.

haslinghuis, 2022-06-06:
In the long term blockchain technology could solve it 😛

freddy22frog, 2022-06-19:
I think having your currently loaded presets stored and then viewed somehow would be great. 
For individuals with a lot or just one Quad, the ability to view your current Presets would be nice just to remember if you loaded UAV Tech or Bardwell Rates, Tunes or Filters, Etc.

limonspb, 2022-06-20:
i think the limiting factor would be space on the FC, so you will have to store the ID of every applied preset in the array.
Besides the ID you want to store the link to the custom preset repository if the user used it...

closed 2022-07-29 by github-actions[bot]