betaflight/betaflight-configurator #2894: Issues with Diatone H743 flight stack

lukedavid2, 2022-04-14:
### Describe the bug

Despite reflashing, the following errors are persistent for the Diatone H743 on Betaflight 4.3 RC4. Could be user error but unable to find other instances of these issues. 

B1: i was able to install Betaflight 4.3 RC4 on my newer 2018 mac, but not on the 2015 one. I have built other drones and flashed them using this laptop, so I would assume it has something to do with the new configurator/ Betaflight 4.3 RC4 and might be worth investigating. it appears that the computer loses the connection with the flight controller partway through the erase/ flash process, and then a new connection can only be initiated by placing the flight controller in boot mode. 

B2: Betaflight menu OSD - Once in the  OSD betaflight menu, settings can be updated just fine, but when it comes to the exit and save menu - the normal command right stick deflection doesn't work well. It seems to cause the drone to freeze and can even accidentally open the camera menu. 

B3: Blackbox - under the black-box tab, the Diatone H743 stack shows a non-existent SD card slot 

B4: I haven't been able to access the blackbox logs created on BF 4.3 RC4, I activated USB mass storage device but the FC freezes before it reappears in finder

### To Reproduce

B1: Used a 2015 mac and the latest version of the configurator that compliments Betaflight 4.3 RC4. Flashing a Diatone H743 with dual MPU6000 gyros - [Here](https://uk.banggood.com/30_5x30_5mm-MAMBA-MK4-H743-3-6S-Flight-Controller-BMI270-or-MPU-6-Series-DUAL-Bulit-in-BEC-for-RC-Drone-FPV-Racing-p-1929033.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=GB&utm_content=minha&utm_campaign=minha-gbg-en-pc&currency=GBP&cur_warehouse=CN&createTmp=1&ID=6317293&utm_source=googleshopping&utm_medium=cpc_us&utm_campaign=jeff-co-pmax-ukbg-allcat-uk-220412&utm_content=jeff&ad_id=&gclid=Cj0KCQjwjN-SBhCkARIsACsrBz7sAxU4PrWpIJwluZGw4CAc1_xSCStZWHOI1NqIhdv8Y7ZOh0H27FQaAhflEALw_wcB)

B2: Simply change a setting in the OSD Betaflight and attempt to save this out and return to the normal flight osd view and close the menu. 

B3: Try to access black-box logs using the normal method through the configurator. Opened using mac os. 

B4 & B5: I haven't been able to access the blackbox logs created on BF 4.3 RC4, I activated USB mass storage device but the FC freezes before it reappears in finder

### Expected behavior

B1: normal flashing procedure. 
B2: Save and exit Betaflight OSD - simplified turning. 
B3: No SD card slot visible, as no sd card slot installed. 
B4: Open mass storage mode and provide access to black-box logs. 

### Configurator version

10.8.0-debug-04b4c74

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
# 

# diff all

# version
# Betaflight / STM32H743 (SH74) 4.3.0 Apr  6 2022 / 07:58:58 (2fd2e8eca) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAH743, version: 48ff4861, date: 2022-04-03T17:41:53Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAH743
manufacturer_id DIAT
mcu_id 0033002b3230510a33343930
signature 

# feature
feature -RX_PARALLEL_PWM
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature DISPLAY

# serial
serial 2 1 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 12,12::C:1
led 1 13,12::C:1
led 2 14,12::C:1
led 3 15,12::C:1
led 4 12,13::C:1
led 5 13,13::C:1
led 6 14,13::C:1
led 7 15,13::C:1
led 8 12,14::C:1
led 9 13,14::C:1
led 10 14,14::C:1
led 11 15,14::C:1
led 12 12,15::C:1
led 13 13,15::C:1
led 14 14,15::C:1
led 15 15,15::C:1

# aux
aux 0 0 0 1850 2100 0 0
aux 1 13 3 1700 2100 0 0
aux 2 36 3 1700 2100 0 0

# rxrange
rxrange 1 1000 1992

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 5
vtxtable powervalues 14 23 27 30 0
vtxtable powerlabels 25 200 500 MAX 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set gyro_to_use = BOTH
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = 62,0,-42,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set ibata_scale = 80
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set osd_ah_pos = 14606
set osd_warnings_pos = 14825
set osd_avg_cell_voltage_pos = 6624
set osd_efficiency_pos = 208
set osd_displayport_device = MSP
set debug_mode = DUAL_GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5658
set displayport_msp_serial = 2

profile 0

profile 1

# profile 1
set dterm_lpf1_dyn_min_hz = 127
set dterm_lpf1_dyn_max_hz = 255
set dterm_lpf1_dyn_expo = 8
set dterm_lpf1_type = BIQUAD
set dterm_lpf1_static_hz = 0
set dterm_lpf2_static_hz = 0
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set p_pitch = 82
set i_pitch = 221
set d_pitch = 54
set f_pitch = 0
set p_roll = 79
set i_roll = 211
set d_roll = 47
set f_roll = 0
set p_yaw = 79
set i_yaw = 211
set f_yaw = 0
set d_min_roll = 47
set d_min_pitch = 54
set simplified_master_multiplier = 160
set simplified_i_gain = 150
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 0
set simplified_dterm_filter_multiplier = 170

profile 2

# restore original profile selection
profile 1

rateprofile 0

# rateprofile 0
set thr_expo = 61

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

haslinghuis, 2022-04-23:
Please provide the output of `resource show all`

You might want to try: https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20220423.99 and test the latest nightly firmware build.
 
B1. Please provide specifics about the 2015 and 2018 mac (CPU and OS version).
B3. Can be fixed by `set sdcard_mode = OFF` (same with Holybro Kakute H7 Mini: https://github.com/betaflight/betaflight/issues/11440).

lukedavid2, 2022-04-23:
Hi Haslinghuis, 

Thanks for your response - I have pasted the **resource show all** dump below BF4.3 RC5

Ill try the nightly build posted and report back. 

B1: 
2015: Macos Monterey 12.3. Macbook pro 13 inch 2015 2.7ghz Dual core I5 8Gb RAM 
2018: Macod Bigsur 11.6.5. Macbook pro 15 inch 2018  2.2ghz I7 6-core I7 16gb RAM

B3: Thank you - this worked perfectly

Thanks,
Luke

# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: SPI_MOSI 3
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: PINIO 1
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: SERIAL_TX 4
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: BEEPER
E04: LED 2
E05: LED 1
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: GYRO_CS 2
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: DSHOT_BITBANG 1
DMA1 Stream 1: SPI_MOSI 1
DMA1 Stream 2: SPI_MISO 1
DMA1 Stream 3: SPI_MOSI 3
DMA1 Stream 4: SPI_MISO 3
DMA1 Stream 5: SPI_MOSI 4
DMA1 Stream 6: SPI_MISO 4
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE

#

haslinghuis, 2022-12-02:
Fixed (https://github.com/betaflight/betaflight-configurator/issues/2894#issuecomment-1107523357)

closed 2022-12-02 by haslinghuis