betaflight/betaflight-configurator #2898: Failsafe time values showed 10x smaller in RC5 than in 10.7.2

LoakAudio, 2022-04-17:
### Describe the bug

On Failsafe tab, the 2 times of the Stage 2 parameters ("Wait for stage 2" and the next under ... Sorry my BF is in french) show 10x smaller than the same quad in BF configurator 10.7.2.

Inversely, a value setted in RC5 will be 10x bigger in RC5.

### To Reproduce

- Config the Failsafe tab for a quad with BF 4.2.11 in the 10.7.2 configurator
- Watch the values in the RC5 configurator.

### Expected behavior

Expetcted to have the same values

### Configurator version

RC5 - fa662d2

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-04-17:
The unit in UI has changed. Please verify the cli parameter on both firmware versions. They should be equal.

LoakAudio, 2022-04-17:
Ok, as the UI still says [1 = 0.1 sec] I was confused

haslinghuis, 2022-04-17:
I guess the translations are not up to date yet.

4.3: 
[image]

4.2: 
[image]

haslinghuis, 2022-04-19:
Fixed in #2900

closed 2022-04-23 by blckmn