betaflight/betaflight-configurator #2905: IOS Compatibility

Shadow22exe, 2022-04-24:
### Is your feature request related to a problem? Please describe

I would like to configure my drone by other cable with my iOS iPhone 8 so I don’t have to come home to change just a setting 

### Describe the solution you'd like

Be able to install the beta flight on iOS and be able to change setting like on android or windows 

### Describe alternatives you've considered

I use my old buggy android but it isn’t convenient

homepods, 2022-05-13:
I agree with you as now more Apple devices have USB-c and are compatible with OTG and can power 5v Fc.

I have tried to run the current configurator to build it on iPadOS through an IDE with node.js but there is a low code which requests git and cannot install git systemwide. 

Although betaflight configurator is JS should be able to run through IDE on iOS.

closed 2022-06-21 by github-actions[bot]

jappyjan, 2023-08-28:
i would like to reopen this.

since a while we now have elrs wifi option which enables us to configure betaflight wirelessly. this opens up a new possibility to use betaflight on ios even without full serial port functionality.