betaflight/betaflight-configurator #2911: Hang after Load Firmware [Local]

howels, 2022-05-02:
### Describe the bug

UI stops responding after loading local firmware on RC6.

On console these errors are seen:
```
[188116:188125:0502/124637.969979:ERROR:login_database.cc(704)] Password store database is too new, kCurrentVersionNumber=29, GetCompatibleVersionNumber=31
[188116:188126:0502/124637.969981:ERROR:login_database.cc(704)] Password store database is too new, kCurrentVersionNumber=29, GetCompatibleVersionNumber=31
[188116:188125:0502/124637.970048:ERROR:password_store_impl.cc(56)] Could not create/open login database.
[188116:188126:0502/124637.970096:ERROR:password_store_impl.cc(56)] Could not create/open login database.
[188145:188145:0502/124638.060735:ERROR:shared_context_state.cc(73)] Skia shader compilation error
------------------------

Errors:
Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 16 2022. Please check build information of source that generated the binary.
[188145:188145:0502/124638.356913:ERROR:shared_context_state.cc(73)] Skia shader compilation error
------------------------

Errors:
Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 16 2022. Please check build information of source that generated the binary.
[188145:188145:0502/124638.358798:ERROR:shared_context_state.cc(73)] Skia shader compilation error
------------------------

Errors:
Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 16 2022. Please check build information of source that generated the binary.
[188145:188145:0502/124638.490959:ERROR:shared_context_state.cc(73)] Skia shader compilation error
------------------------

Errors:
Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 16 2022. Please check build information of source that generated the binary.
[188116:188139:0502/124707.980522:ERROR:database.cc(1709)] History sqlite error 1, errno 0: no such column: publicly_routable, sql: SELECT id,url,visit_time,from_visit,transition,segment_id,visit_duration,incremented_omnibox_typed_score,publicly_routable FROM visits WHERE visit_time >= ? AND visit_time < ?ORDER BY visit_time LIMIT ?
```

### To Reproduce

Load local firmware with board connected, UI stops responding.

### Expected behavior

Firmware file is loaded and can click "Flash Firmware" button.

### Configurator version

be4a9e1 (4.3.0 RC6)

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

### Add any other context about the problem that you think might be relevant here

OS is Ubuntu Linux 22.04

haslinghuis, 2022-05-02:
Cannot reproduce using Ubuntu 20.04 LTS.

Did you select a unified or local config before flashing the local hex?

https://askubuntu.com/questions/1360148/skia-shader-compilation-error

howels, 2022-05-02:
> Cannot reproduce using Ubuntu 20.04 LTS.
> 
> Did you select a unified or local config before flashing the local hex?
> 
> https://askubuntu.com/questions/1360148/skia-shader-compilation-error

Bug only appeared after upgrade to 22.04, likely related to libraries.

I am selecting a board with a unified config and a local hex for e.g. STM32H743 or STM32F7X2. 

Is it possible to obtain more verbose output on the CLI?

howels, 2022-05-02:
The error "Skia shader compilation error" is not relevant. I can run with `--disable-gpu` and that solves the skia error but the same behaviour exists - hangs when loading firmware.

Remaining errors seem to talk about a DB... possible an sqlite instance in here somewhere?
```
[195199:195221:0502/143901.887785:ERROR:login_database.cc(704)] Password store database is too new, kCurrentVersionNumber=29, GetCompatibleVersionNumber=31
[195199:195209:0502/143901.887813:ERROR:login_database.cc(704)] Password store database is too new, kCurrentVersionNumber=29, GetCompatibleVersionNumber=31
[195199:195221:0502/143901.887844:ERROR:password_store_impl.cc(56)] Could not create/open login database.
[195199:195209:0502/143901.887877:ERROR:password_store_impl.cc(56)] Could not create/open login database.
[195230:195238:0502/143927.964078:ERROR:nss_util.cc(286)] After loading Root Certs, loaded==false: NSS error code: -8018
[195230:195238:0502/143927.964432:ERROR:cert_verify_proc_builtin.cc(559)] No net_fetcher for performing AIA chasing.
[195199:195285:0502/143931.886593:ERROR:database.cc(1709)] History sqlite error 1, errno 0: no such column: publicly_routable, sql: SELECT id,url,visit_time,from_visit,transition,segment_id,visit_duration,incremented_omnibox_typed_score,publicly_routable FROM visits WHERE visit_time >= ? AND visit_time < ?ORDER BY visit_time LIMIT ?
[195199:195285:0502/144001.900475:ERROR:database.cc(1709)] History sqlite error 1, errno 0: no such column: publicly_routable, sql: SELECT id,url,visit_time,from_visit,transition,segment_id,visit_duration,incremented_omnibox_typed_score,publicly_routable FROM visits WHERE visit_time >= ? AND visit_time < ? AND (transition & ?) == ?ORDER BY visit_time LIMIT ?
```

howels, 2022-05-02:
Cleared ~/.config/betaflight-configurator and ~/.cache/betaflight-configurator.  Messages are gone (bad DB replaced) but error persists.

haslinghuis, 2022-05-02:
Did you follow the steps here: https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md#flashing-a-target-with-betaflight-configurator-on-ubuntu-2004

howels, 2022-05-02:
> Did you follow the steps here: https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md#flashing-a-target-with-betaflight-configurator-on-ubuntu-2004

Yes, of course I followed the steps and set the permissions, I have done this before. This is some new strangeness that isn't present on my other laptop, and wasn't present prior to upgrade. My /etc/udev/rules.d is the same on this broken system as on my working system. I can also work in configurator fine up until I load local firmware. I cannot even hit the flash button to trigger DFU. If I enter to DFU prior to loading the flash screen I see the same lockup - as soon as I load local firmware.

I have added logging in the hope that it explains the problem.

[bg-configurator-log.txt](https://github.com/betaflight/betaflight-configurator/files/8603424/bg-configurator-log.txt)

howels, 2022-05-02:
FWIW this is a fresh 22.04 install with only 2 udev rules set for STM32 and stLinkv2 devices:

```
$ cat /etc/udev/rules.d/45-stdfu-permissions.rules 
# DFU (Internal bootloader for STM32 MCUs) Stephen
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"
# non-DFU mode
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE="0664", GROUP="plugdev"

$ cat /etc/udev/rules.d/49-stlinkv2.rules 
# stm32 discovery boards, with onboard st/linkv2
# ie, STM32L, STM32F4.
# STM32VL has st/linkv1, which is quite different

SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="3748", \
    GROUP:="dialout", \
    SYMLINK+="stlinkv2_%n"
```
I kept my profile but the rest of the install is fresh with no old libraries at all.

haslinghuis, 2022-05-02:
Will check later on another machine with Ubuntu 22.04 LTS.

haslinghuis, 2022-05-02:
OK, works on Ubuntu 20.04 LTS with NW.js 0.54. Ubuntu 22.04 LTS requires NW.js 0.60 to solve the flashing issue locking up configurator.

Unfortunately we can't upgrade because of this issue: https://github.com/betaflight/betaflight-configurator/pull/2856 😞 

So somebody has to solve the OSD issue to make it work for both.

haslinghuis, 2022-05-02:
Another pitfall on Ubuntu 22.04 LTS is it's using `clang` v14 as firmware is not compatible yet (need v7 - v13) used for `make test`

haslinghuis, 2022-05-03:
@howels please check this build: https://github.com/betaflight/betaflight-configurator/pull/2913#issuecomment-1115557255

howels, 2022-05-03:
> @howels please check this build: [#2913 (comment)](https://github.com/betaflight/betaflight-configurator/pull/2913#issuecomment-1115557255)

Confirm that https://github.com/betaflight/betaflight-configurator/pull/2913 fixes the issue. Flashing works normally now. Thanks!

mrRobot62, 2022-05-11:
Tested this issue with latest nightly under MacOS (Mojave) without any problems. Did this with several differnent FC and for every FC several times.

PR #2913 (2022-05-19): Fix freezing of configurator while loading local firmware and OSD preview while moving elements on Linux
Fixes: 
#2911 
#2878

- [x] Update workflow to use Ubuntu 22.04 LTS (needed to make this PR build)
- [x] Update Node to 16.x. superseeds https://github.com/betaflight/betaflight-configurator/pull/2782
- [x] Update NW.js to 0.62 (synchronized versions on all platforms)
- [x] Remove OSD dragging preview on linux (as this was an issue after 0.54.1)

Reminder: Update ci.yml as Ubuntu 22.04 is released.
closed 2022-05-19 by haslinghuis