betaflight/betaflight-configurator #2918: Emergency motor stop

JimB40, 2022-05-09:
### Is your feature request related to a problem? Please describe

No quick and safe way to stop motors once testing in Motor tab

### Describe the solution you'd like

[image]

### Describe alternatives you've considered

As discussed in discord with @limonspb.
Big, easily clickable button plus shorcut to biggest keyboard key which is Space Bar
Palced in UI so it cannot be hidden while window is resized

limonspb, 2022-05-09:
I think the button should be close to the motor sliders. For example check android version:
[image]

haslinghuis, 2022-05-12:
@JimB40 please check #2921 - no need for UI changes here - added spacebar handler. Just needs documentation and mentioning  in youtube guides.

haslinghuis, 2022-05-17:
@JimB40 check again. Added a simple button.

PR #2921 (2022-05-28): Stop motor test with Any key
Fixes: #2918

- [x] Add `ANY` key handler to stop motor test.
- [x] Add `Stop Motors` button. 
- [x] Only poll MSP status updates when testing is enabled.
- [x] Updates MSP promises to `async await`.
closed 2022-05-28 by blckmn