betaflight/betaflight-configurator #2924: set usb_hid_cdc=on breaks ability to connect until firmware reflashed

dkjroot, 2022-05-14:
### Describe the bug

On betaflight 10.7.2 and firmware 4.2.11 (GEPRCF411), usb_hid_cdc mode was working fine.  Since flashing 4.3.0-rc6 of the firmware (GEPRCF411) and upgrading to 10.8.0.rc6 of betaflight, if I set usb_hid_cdc=ON then I cannot connect to betaflight until I reflash the FC.  This happens consistently, and with no modifications from the custom defaults that come with the firmware.

- Driver of Betaflight STM32F411 Interface 1 is WinUSB (v6.1.7600.16385) (I have also tried this with libusb-win32 (v1.2.6.0))

- Driver of Betaflight STM32F311 Interface 0 is HidUsb (v10.0.19041.868)

- OS is Windows 10 x64

### To Reproduce

set usb_hid_cdc=on; save; 
After doing that, drone reboots and I cannot connect any more ("Failed to open serial port").

### Expected behavior

Connect should work and the receiver should work as a game controller.

### Configurator version

10.8.0 rc6

### Flight controller configuration

```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.3.0 Apr 25 2022 / 01:03:20 (9d71184) MSP API: 1.44
# config: manufacturer_id: GEPR, board_name: GEPRCF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name GEPRCF411
manufacturer_id GEPR
mcu_id 003e00303030511737393838
signature 

# master
set acc_calibration = 82,62,172,1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

haslinghuis, 2022-05-14:
Please enable in options tab:

[image]

haslinghuis, 2022-05-15:
Duplicate of https://github.com/betaflight/betaflight-configurator/issues/2859

closed 2022-05-15 by haslinghuis

dkjroot, 2022-05-15:
Thank you @haslinghuis - I haven’t tried the solution yet but that does look like the same issue and so I’m sure it will work.  Sorry for the duplicate!