betaflight/betaflight-configurator #2925: Sometimes I get stuck on PID Tuning page until I disconnect and reconnect

dkjroot, 2022-05-14:
### Describe the bug

Sometimes when I visit the PID Tuning page, I can't select any other page until I disconnect and reconnect the FC.  I don't need to have made any changes in order to get stuck.

### To Reproduce

Go to the PID Tuning page, sometimes I get stuck there.  I haven't identified what causes it to stick, it's intermittent.

### Expected behavior

Should be able to move onto another page if unchanged or after saving changes to PID Tuning

### Configurator version

10.8.0.RC6

### Flight controller configuration

```
# version
# Betaflight / STM32F411 (S411) 4.3.0 Apr 25 2022 / 01:03:20 (9d71184) MSP API: 1.44
# config: manufacturer_id: GEPR, board_name: GEPRCF411, version: 7b156dec, date: 2021-07-26T13:19:23Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name GEPRCF411
manufacturer_id GEPR
mcu_id 003e00303030511737393838
signature 

# name: Buddy!

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0

# feature
feature -AIRMODE
feature -ANTI_GRAVITY
feature LED_STRIP

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 2,4::CTO:6
led 1 3,4::CTO:6
led 2 4,4::CTO:6
led 3 5,4::CTO:6
led 4 6,4::CTO:6
led 5 7,4::CTO:6
led 6 8,4::CTO:6
led 7 9,4::CTO:6
led 8 10,4::CTO:6
led 9 11,4::CTO:6
led 10 12,4::CTO:2
led 11 13,4::CTO:2

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 4 1 900 1700 0 0
aux 4 13 2 1300 1700 0 0
aux 5 28 1 900 1700 0 0
aux 6 35 2 1700 2100 0 0
aux 7 47 1 900 1300 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 700
set gyro_lpf1_dyn_min_hz = 350
set gyro_lpf1_dyn_max_hz = 700
set acc_calibration = 22,-61,223,1
set dshot_idle_value = 800
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 180
set vbat_max_cell_voltage = 435
set vbat_warning_cell_voltage = 340
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set simplified_gyro_filter_multiplier = 140
set osd_vbat_pos = 375
set osd_flymode_pos = 385
set osd_compass_bar_pos = 362
set osd_altitude_pos = 83
set osd_warnings_pos = 14377
set osd_avg_cell_voltage_pos = 2422
set osd_battery_usage_pos = 392
set osd_nheading_pos = 76
set osd_up_down_reference_pos = 238
set osd_rate_profile_name_pos = 55
set vtx_band = 4
set vtx_channel = 5
set vtx_power = 2
set vtx_freq = 5820
set name = Buddy!

profile 0

# profile 0
set anti_gravity_gain = 0

profile 1

# profile 1
set anti_gravity_gain = 0

profile 2

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set thr_expo = 47
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_breakpoint = 1650

rateprofile 1

# rateprofile 1
set thr_expo = 39
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 57
set pitch_srate = 57
set yaw_srate = 57
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

rateprofile 2

# rateprofile 2
set thr_expo = 39
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 45
set pitch_srate = 45
set yaw_srate = 45
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 3

# rateprofile 3
set thr_expo = 39
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 57
set pitch_srate = 57
set yaw_srate = 57
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

rateprofile 4

# rateprofile 4
set thr_expo = 39
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 45
set pitch_srate = 45
set yaw_srate = 45
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
```

haslinghuis, 2022-05-15:
Please provide OS and version used.

Please try builds in: 

1. https://github.com/betaflight/betaflight-configurator/pull/2913
2. https://github.com/betaflight/betaflight-configurator/pull/2846.

On both we have console available for having further info (using F12)

dkjroot, 2022-05-15:
Sorry, I thought I said - Windows 10 x64, 10.8.0-rc6.  I’ll check out those two threads and let you know if it sheds any light

closed 2022-06-22 by github-actions[bot]