betaflight/betaflight-configurator #2933: Unable to install betaflight-configurator on ubuntu: The following packages have unmet dependencies on Ubuntu 22.04LTS

Aeromarine, 2022-05-23:
### Describe the bug

after download the https://github.com/betaflight/betaflight-configurator/releases/download/10.8.0-RC6/betaflight-configurator_10.8.0_amd64.deb install libatomic1, then install .deb file from the link I get this error https://imgur.com/HgwfoUq

what have I done wrong? I have tried everythin and searched the Google for the passed week.. still have find the answer.. please help..

### To Reproduce

what I did: fresh install ubuntu 22.04LTS
then: sudo apt install libatomic1 in terminal 
then: install https://github.com/betaflight/betaflight-configurator/releases/download/10.8.0-RC6/betaflight-configurator_10.8.0_amd64.deb

then the error..

### Expected behavior

no idea what I should do from here ...

### Configurator version

Betaflight Configurator 10.8.0 Release Candidate 6

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-05-23:
Support for 22.04LTS was added here. Have to check my machine if I needed libatomic or not.

Aeromarine, 2022-05-23:
thank you for the reply. and just to be sure. doest Betaflight-Configurator support Arm64 architecture? I am using it on RPi...

haslinghuis, 2022-05-23:
Yes I'd installed libatomic1 on Ubuntu 22.04LTS. It's supports AMD64 arch.

You also need: https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Ubuntu.md#flashing-a-target-with-betaflight-configurator-on-ubuntu-2004

McGiverGim, 2022-05-23:
If this library is needed, I think maybe we need to include it here?

https://github.com/betaflight/betaflight-configurator/blob/d92a49aac8a1c079e812cb7e361c4383a460010b/gulpfile.js#L731

McGiverGim, 2022-05-23:
I think remember the depends is a list of dependencies needed by the debian package.

McGiverGim, 2022-05-23:
Strange, looking at the documentation of the gulp-debian package, the depends parameter does not exist. But I think remember I tested it and worked when the code was created...

closed 2022-05-28 by blckmn