betaflight/betaflight-configurator #2935: The license in Betaflight Configurator appears as propietary

McGiverGim, 2022-05-24:
### Describe the bug

When looking at the package in Ubuntu, it is seen as propietary:

[image]

### To Reproduce

Open the deb package with the software installer.

### Expected behavior

Appear GPL as license, not propietary.

### Configurator version

Not important

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-05-24:
Do we need op copyrights file?
https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/

haslinghuis, 2022-05-24:
Do we need a copyrights file?
https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/

McGiverGim, 2022-05-24:
In theory is added, and I've seen others with the same problem in internet, but I've not found a workaround. I've opened the issue to see if someone with more knowledge in Linux finds something.

haslinghuis, 2024-05-15:
Now we are moving to PWA this would no longer be necessary

closed 2024-05-15 by haslinghuis