betaflight/betaflight-configurator #2938: Massive problems in the firmware flasher tab

TheIsotopes, 2022-05-29:
### Describe the bug

Since yesterday's merge, I've had massive problems with the Flasher tab. when I go to flashing this image appears (see screenshot). After switching through the tabs and restarting several times, it seems to work again. but only once, after a restart the flasher tab is defective again.

here the screenshot and console output:
[image]
[image]

### To Reproduce

Using latest master and going to Firmware flasher Tab.

### Expected behavior

working flashing tab :(

### Configurator version

10.8.0-debug-10645cb7

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-05-29:
Thanks for reporting. Please test https://github.com/betaflight/betaflight-configurator/pull/2937.
Will look into this today.

TheIsotopes, 2022-05-29:
> Please test https://github.com/betaflight/betaflight-configurator/pull/2937

Nope, unfortunately not.

The sledgehammer method of deleting the `C:\Users\xyz\AppData\Local\betaflight-configurator` folder is still the fastest and easiest way.

haslinghuis, 2022-05-29:
@TheIsotopes. Updated a new version.

Quota has been exceeded. Have moved firmware flasher storage to session storage.

TheIsotopes, 2022-05-30:
Yes, everything is working as it should now!!

Thx @haslinghuis 👍

PR #2937 (2022-05-31): Fix MSP_BOARD_INFO accumulation and localStorage Quota Exceeded
Fixes: #2938 

- [x] Fixes `FC.CONFIG.boardName` parameter accumulating previous values.
- [x] Move `Loader` function in a `try catch error` statement.
- [x] Move `Jenkins` + firmware `cache` from `localStorage` to new `SessionStorage` to fix `Exceeded Quota` error.
- [x] Remove old keys from LocalStorage.
closed 2022-05-31 by Asizon