betaflight/betaflight-configurator #2939: Web Based Node.js Betaflight Configurator

homepods, 2022-05-29:
### Is your feature request related to a problem? Please describe

We need to make a solution which covers the internet available mobility and customisation of FPV flyers and install a one off donation and for a SaaS Betaflight configurator accessible from anywhere anytime or with instant availability.
As new USB-C standard emerges in Europe and with speedybee monopolising the device connectivity while selling open source code to buy their usb wireless connectivity I feel the need to accommodate mobile first approach of the betaflight configurator.

### Describe the solution you'd like

I am investing in domain name and VM hosting and I need support to access USB via web in non chrome virtual environments on Linux machine. 

I can deploy the web server and I need support with reading USB port input tty or drive 

if as when you go to BFC connect your device like in 😜🖲 (https://esc-configurator.com/) but can access from Mozilla , WebKit and Chromium. 

### Describe alternatives you've considered

Other option is to hire and independent data encryption and storage from a public provider or to receive a donation from azure web services or aws or Google cloud to encrypt at rest. 

Perhaps we should aggregate and support mobile first approach. 

### Other information

We are looking to improve flight experience beyond one’s imagination

FiercestT, 2022-05-29:
That's a shit ton of buzzwords. Seems like a pitch to a company rather than an open source project.

homepods, 2022-05-31:
One might argue that a browser based betaflight configurator through api serial ☎️ chrome edge opera will improve operability and reduce crashes as the version will be optimised for one machine

haslinghuis, 2022-05-31:
Last notes: https://github.com/betaflight/betaflight-configurator/pull/2263#issuecomment-728242790

I'd like a web-based configurator with CI/CD - we just need to migrate API's (anyway).

haslinghuis, 2024-05-06:
Checkout https://configurator.betaflight.com. It's still in early development as flashing is not yet possible,

closed 2024-05-06 by haslinghuis