betaflight/betaflight-configurator #2947: Motor test. Maximum amplitude.

ekap12, 2022-06-13:
### Is your feature request related to a problem? Please describe

In the motors tab on the gyroscope graph, weak and high-frequency vibrations are not displayed. Or at least their maximum amplitude.

### Describe the solution you'd like

Is it possible to get the maximum amplitude of motor vibrations not from low-frequency data, but directly from the flight controller?
When I rotate the motors in batafly, it does not display the maximum vibration amplitude, apparently due to the too low frequency of data transfer to the configurator, although the blackbox log clearly shows that there are vibrations. Counting (and the ability to reset or, say, transmitting a maximum in 1 second) the maximum amplitude on the controller side would solve this problem and it would be possible to accurately balance the motors and propellers from the configurator.

Now I can clearly see which of my motors has a higher vibration only by looking at the logs, but the configurator does not display this in any way.

### Describe alternatives you've considered

1. Receive and output data from the gyroscope with a higher frequency but I think that the current speed is limited by the data rate.
2. To transmit the maximum amplitude from the controller since the last press of the reset button.
3. Transfer the maximum amplitude once a second for example and draw it on a graph.

### Other information

[image]

ekap12, 2022-06-13:
> * You can look at the gyro traces in the plot in the motors tab, if I remember correctly

Yes, but unfortunately these graphs do not display correct information if the noise is at a high frequency.
You can compare two graphs, one from the configurator (you can't see the motor noise on it at all) and one from the blackbox. Blackbox I will also attach.
[image]
[image]
[LOG00112.zip](https://github.com/betaflight/betaflight-configurator/files/8891935/LOG00112.zip)

KarateBrot, 2022-06-13:
Look at the RMS values (in brackets). It's a measure of the noise (if you don't touch the quad to move it around).

[image]

ekap12, 2022-06-13:
Thanks for answers! All I understood is what the problem is, apparently the data in these graphs are taken from the hyoscope after filtering. Now I have disabled all filters and it seems to work as it should. Perhaps it is better to immediately take unfiltered data.

haslinghuis, 2022-06-13:
Please in cli do
```
set debug_mode = `gyro_raw`
save
```
and check sensor tab

ekap12, 2022-06-13:
> and check sensor tab
Before that, I had gyro_scaled . I made a gyro_raw everything is fine without filters, but if you turn on the filters, then the maximum amplitude drops dramatically and the graph is as smooth as it was.
In the sensors tab including.

haslinghuis, 2022-06-13:
Look at the debug mode in configurator instead to see the raw gyro sensor values

haslinghuis, 2022-06-13:
Enable motor testing in motors tab, select your throttle value and go to sensors tab

ekap12, 2022-06-13:
I made a button in the configurator which is:
1) spins the motors to the speed I need
2) waits 0.2 seconds.
3) Press the button to reset the maximum amplitudes (RESET)
4) within 1 second waiting (waiting until the maximum amplitude is caught on each axis)
5) record the maximum amplitude of the data that are in brackets.
6) stop the motor.

This, with very good repeatability, gives me an indication of how badly the motor is out of balance.
it would be great if this was implemented immediately in the configurator, unfortunately my implementation is not suitable for public distribution.

And now, after disabling the filters, it all works very well.

ekap12, 2022-06-13:
> Enable motor testing in motors tab, select your throttle value and go to sensors tab

I didn't think there was such a possibility. I honestly did not expect that the data in the motors tab could be filtered.
I think that for myself I will just replace FC.SENSOR_DATA.gyroscope[....] in the file motors.js with FC.SENSOR_DATA.debug[...] with and I will see unfiltered data in the motors tab. Thanks for the help!

haslinghuis, 2022-06-13:
No please don't look at the graph in motors tab but the debug values in sensor tabs instead.

closed 2022-06-18 by ekap12