betaflight/betaflight-configurator #2948: Betaflight 10.8.0 APK outdated bug

TylerDD4, 2022-06-14:
### Describe the bug

After updating to the latest release of 10.8.0 APK it alerts that the Configurator is outdated and tells you how to update. Figured it's a bug, because it does it all the time. 

### To Reproduce

Open betaflight Configurator APK after downloading newest release. 

### Expected behavior

No error message about outdated firmware 

### Configurator version

10.8.0 5ec9465

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

TylerDD4, 2022-06-14:
[image]

TylerDD4, 2022-06-14:
[image]

haslinghuis, 2022-06-15:
FIxed in https://github.com/betaflight/betaflight-configurator/pull/2942

closed 2022-06-15 by haslinghuis

jawz101, 2022-07-01:
betaflight-configurator_10.8.0-debug-5ec9465_android.apk

Does this version have the fix?  Because I'm still getting the error on this one

TylerDD4, 2022-07-01:
> betaflight-configurator_10.8.0-debug-5ec9465_android.apk
> 
> Does this version have the fix? Because I'm still getting the error on this one

Exactly... Replied as fixed, but the error hasn't gone away.

jawz101, 2022-07-01:
The above dude posted a commit that changed the build ID for Betaflight from 10.8 to 10.9 so I have no idea how that means Android app's error goes away for v10.8.