betaflight/betaflight-configurator #2949: 10.8.0(5ec9465) Can't find my serial port, which is definitely exist.

lida2003, 2022-06-14:
### Describe the bug

10.8.0(5ec9465) Can't find my serial port, which is definitely exist and ok with previous 10.7.2(without pc reboot).

### To Reproduce

1) launch 10.8.0(5ec9465) 
2) plug fc serial port
3) I didn't see serial port in drop down list
4) windows manager show COM5 existed and OK

### Expected behavior

just same as 10.7.2 did, can  auto detect my fc serial port.

### Configurator version

10.8.0(5ec9465)

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```
I' think it not relevant to FC.

### Add any other context about the problem that you think might be relevant here

do I need reboot pc when upgrade the configurator?

lida2003, 2022-06-14:
Reboot PC can't help. Fallback to 10.7.2, It's OK.

haslinghuis, 2022-06-14:
Please show screenshot of (windows-R) devmgmt.msc in Windows - Ports (COM & LPT) when FC is attached

lida2003, 2022-06-15:
[image]
[image]
[image]

haslinghuis, 2022-06-15:
Two options to try:

Enable show all serial devices (option tab while disconnected):
[image]

Revert to english (welcome page while disconnected):
[image]

lida2003, 2022-06-15:
Configurator is using english as default.
[image]

haslinghuis, 2022-06-15:
Please try a nightly build so we can look at dev console using F12 when you plug in we should see something happening:
https://github.com/betaflight/betaflight-configurator/pull/2945#issuecomment-1148711357

lida2003, 2022-06-15:
Seems work for me!  10.9.0-debug-c67360c
[image]

closed 2022-06-15 by lida2003