betaflight/betaflight-configurator #2950: Configurator on Fedora 36 Workstation

zandorsp, 2022-06-15:
### Describe the bug

I'm tried to install BetaFlight configurator 10.8.0 on Fedora 36, but it have a dependency unsolved: libgconf-2-4

### To Reproduce

zandor@tiger Downloads]$ sudo dnf install ./betaflight-configurator-10.8.0-1.x86_64.rpm 
[sudo] password for zandor: 
Last metadata expiration check: 0:02:44 ago on Tue 14 Jun 2022 11:00:49 PM -03.
Error: 
 Problem: conflicting requests
  - nothing provides libgconf-2-4 needed by betaflight-configurator-10.8.0-1.x86_64
(try to add '--skip-broken' to skip uninstallable packages)

### Expected behavior

Successful instalation

### Configurator version

10.8.0

### Flight controller configuration

Do not apply

haslinghuis, 2022-06-15:
$ provides libgconf-2.so.4

haslinghuis, 2022-06-22:
By removing the x32 GConf package and installing the x64 GConf package,

https://linux-packages.com/fedora-36/package/gconf2x86-64

mituritsyn, 2022-06-23:
Faced same issue but all dependecies are installed, force install resolve the issue
rpm --install --nodeps or something like this

alex27riva, 2022-06-23:
Ok, I installed i with `sudo rpm --install --nodeps betaflight-configurator-10.8.0-1.x86_64.rpm`

haslinghuis, 2022-06-23:
Looking but can't find the right package to enter in:
https://github.com/betaflight/betaflight-configurator/blob/863dd3a8c159084bdec83f30ebf8c03d3419d481/gulpfile.js#L772

mituritsyn, 2022-06-23:
probably issue in packet naming convention
```
ls -l /usr/lib64/libgconf*
lrwxrwxrwx. 1 root root     19 янв 26  2021 /usr/lib64/libgconf-2.so -> libgconf-2.so.4.1.5
lrwxrwxrwx. 1 root root     19 янв 26  2021 /usr/lib64/libgconf-2.so.4 -> libgconf-2.so.4.1.5
-rwxr-xr-x. 1 root root 219096 янв 26  2021 /usr/lib64/libgconf-2.so.4.1.5

sudo dnf list installed | grep libatomic
libatomic.i686                                    11.3.1-2.fc34                                 @updates                
libatomic.x86_64                                  11.3.1-2.fc34                                 @updates  
```

haslinghuis, 2022-06-23:
It seems we need the 2nd one in that list, but naming convention is dubious

kevinsumner, 2022-06-28:
I just ran into this on my bench workstation. The dependencies listed in the RPM appear to be package names from Debian (or a descendant distro, like Ubuntu).

kevinsumner, 2022-06-28:
This appears to be it: https://github.com/betaflight/betaflight-configurator/blob/master/gulpfile.js#L772=

kevinsumner, 2022-06-28:
I believe packages should be `GConf2` and `libatomic` in Fedora/RHEL. Not sure about non-Red Hat descendants (SUSE, Mandriva, etc.). Seeing if I can build and install with that locally.

PR #2962 (2022-07-05): Use Fedora/RHEL names for RPM deps
Fixes #2950
closed 2022-07-05 by haslinghuis