betaflight/betaflight-configurator #2957: download links don'r work

S-T-A, 2022-06-24:
### Describe the bug

when i go over the link to try and download it. it's grayed out and there is no link to download.

### To Reproduce

tried diffrent web browsers to no avail i'm on a mac OSX i even updated my lap top same thing is there something im missing
 

### Expected behavior

i thought it would download right away.:(

### Configurator version

beta flight  10.8.0

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-06-24:
Can you describe the link (picture please) you are trying to access?

S-T-A, 2022-06-24:
[](url
[image]
)
i'm tring to downloasd the Mac osx version but non of them highlight as they normally would.

haslinghuis, 2022-06-24:
Please scroll down and you find: https://github.com/betaflight/betaflight-configurator/releases/download/10.8.0/betaflight-configurator_10.8.0_macOS.dmg

closed 2022-06-24 by haslinghuis

S-T-A, 2022-06-24:
Thank you sir you rock!!!!!