betaflight/betaflight-configurator #2958: Dependencies update

McGiverGim, 2022-06-24:
### Is your feature request related to a problem? Please describe

This is only a reminder, with a list of dependencies that can be updated. Some of them, like `three` can't be updated without breaking something (we tried in he past), and others like Vue need a lot of changes, but the major part can be updated, and now is a good moment, with a lot of months of testing before the next release.

This is the list:
```
 @fortawesome/fontawesome-free   ^5.13.0  →    ^6.1.1
 dompurify                        ^2.3.6  →    ^2.3.8
 i18next                         ^19.0.0  →  ^21.8.10
 inflection                      ^1.12.0  →   ^1.13.2
 jbox                             ^1.0.8  →    ^1.3.3
 jquery                           ^3.5.1  →    ^3.6.0
 lru_map                          ^0.3.3  →    ^0.4.1
 marked                          ^4.0.10  →   ^4.0.17
 object-hash                      ^2.0.3  →    ^3.0.0
 semver-min                       ^0.6.5  →    ^0.7.2
 short-unique-id                  ^1.1.1  →    ^4.4.4
 three                           ~0.97.0  →  ~0.141.0
 vue                              2.6.12  →    3.2.37
 @babel/core                     ^7.17.9  →   ^7.18.5
 @rollup/plugin-alias             ^3.1.1  →    ^3.1.9
 @rollup/plugin-commonjs         ^15.1.0  →   ^22.0.0
 @rollup/plugin-node-resolve      ^9.0.0  →   ^13.3.0
 @rollup/plugin-replace           ^2.3.3  →    ^4.0.0
 @storybook/addon-actions        ^6.4.22  →    ^6.5.9
 @storybook/addon-essentials     ^6.4.22  →    ^6.5.9
 @storybook/addon-links          ^6.4.22  →    ^6.5.9
 @storybook/vue                  ^6.4.22  →    ^6.5.9
 babel-loader                     ^8.2.4  →    ^8.2.5
 chai                             ^4.2.0  →    ^4.3.6
 command-exists                   ^1.2.8  →    ^1.2.9
 cordova-lib                     ^10.0.0  →   ^11.0.0
 del                              ^5.0.0  →    ^6.1.1
 eslint                          ^7.16.0  →   ^8.18.0
 eslint-plugin-vue                ^7.3.0  →    ^9.1.1
 follow-redirects                ^1.14.8  →   ^1.15.1
 fs-extra                         ^8.1.0  →   ^10.1.0
 glob                             ^7.1.6  →    ^8.0.3
 gulp-json-editor                 ^2.5.4  →    ^2.5.6
 gulp-replace                     ^1.0.0  →    ^1.1.3
 gulp-xml-transformer             ^5.1.1  →    ^6.0.0
 gulp-zip                         ^5.0.0  →    ^5.1.0
 husky                            ^4.3.0  →    ^8.0.1
 karma                           ^6.3.16  →    ^6.4.0
 karma-chrome-launcher            ^3.0.0  →    ^3.1.1
 karma-mocha                      ^1.3.0  →    ^2.0.1
 karma-rollup-preprocessor        ^7.0.5  →    ^7.0.8
 karma-spec-reporter             ^0.0.32  →   ^0.0.34
 mocha                            ^7.0.1  →   ^10.0.0
 nw-builder                       ^3.5.7  →    ^3.7.4
 os                               ^0.1.1  →    ^0.1.2
 postcss                         ^8.2.13  →   ^8.4.14
 rollup                          ^2.28.2  →   ^2.75.7
 rollup-plugin-vue                ^5.*.*  →    ^6.*.*
 run-script-os                    ^1.1.3  →    ^1.1.6
 simple-git                       ^3.5.0  →    ^3.9.0
 sinon                            ^9.0.0  →   ^14.0.0
 sinon-chai                       ^3.5.0  →    ^3.7.0
 temp                             ^0.9.1  →    ^0.9.4
 vue-loader                      ^15.9.8  →   ^17.0.0
 vue-template-compiler           ^2.6.12  →   ^2.6.14
 yarn                           ^1.22.17  →  ^1.22.19
```

### Describe the solution you'd like

Update all the dependencies that we can.

### Describe alternatives you've considered

Let the old versions like now.

haslinghuis, 2022-06-28:
This list is now reduced to:
```
yarn add gulp-xml-transformer
yarn add husky
yarn add inflection
yarn add lru_map
yarn add three

yarn add rollup-plugin-vue
yarn add eslint-plugin-vue

yarn add vue
yarn add vue-loader
yarn add vue-template-compiler
```

haslinghuis, 2022-06-28:
These dependencies are remaining but need more changes:
```
yarn add gulp-xml-transformer
yarn add husky
yarn add inflection
yarn add lru_map
yarn add three

yarn add rollup-plugin-vue
yarn add eslint-plugin-vue

yarn add vue
yarn add vue-loader
yarn add vue-template-compiler
```

1257mp, 2023-05-29:
Not sure how applicable this can be, but for simple version updates and dependency management, I cannot stop recommending [dependabot](https://github.com/dependabot/dependabot-core). 

It handles _most_ of these types of updates automatically and can make commits documenting each change. It can be set to run on a set interval, making dependency management 100x easier for me. It is worth looking at if you and your team have not already.

haslinghuis, 2024-05-06:
Fixed in #3872

closed 2024-05-06 by haslinghuis