betaflight/betaflight-configurator #2959: Changing font size

oleg-kn, 2022-06-25:
### Is your feature request related to a problem? Please describe

Hello is it possible to change font size in configurator? It's hard to read text on 1440p monitors.

### Describe the solution you'd like

Hello is it possible to change font size in configurator? It's hard to read text on 1440p monitors.

### Describe alternatives you've considered

Hello is it possible to change font size in configurator? It's hard to read text on 1440p monitors.

McGiverGim, 2022-06-25:
It's possible, but maybe it breaks the layout. There is a ton of places where we have sizes in pixels 😞.
Let's see if someone has time to give it a try...

Six6siX, 2024-06-29:
Most browsers have a accessibility feature which will help resize everything in the mean time try

Google Chrome click the three dots in the top right hand corner and look for the Zoom function.
[image]

[https://support.google.com/chrome/answer/96810?hl=en-GB&co=GENIE.Platform%3DDesktop](https://support.google.com/chrome/answer/96810?hl=en-GB&co=GENIE.Platform%3DDesktop)

If using EDGE 
[https://support.microsoft.com/en-gb/microsoft-edge/accessibility-features-in-microsoft-edge-4c696192-338e-9465-b2cd-bd9b698ad19a](https://support.microsoft.com/en-gb/microsoft-edge/accessibility-features-in-microsoft-edge-4c696192-338e-9465-b2cd-bd9b698ad19a)

I've had a quick look at the style sheets, but have come across some unwanted UI breakages. 

[image]