betaflight/betaflight-configurator #2966: Fails to load remote config for any board.

Scout339, 2022-06-30:
### Describe the bug

Failed to load remote config for https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/FLWO-FLYWOOF7DUAL.config

### To Reproduce

go to firmware flasher and select any board, it will say "Loading..." infinitely in the firmware selection dropdown.

### Expected behavior

supposed to load remote config for updating firmware...

### Configurator version

10.8.0

### Flight controller configuration

```
  # REPLACE THIS LINE BY THE OUTPUT OF YOUR `diff`
```

haslinghuis, 2022-06-30:
Checking... Still working.
Please restart configurator and check you internet connection.

Scout339, 2022-06-30:
Restarted PC, going to try it right now. Maybe ill try it on my phone as well to see if it gets the same issue

Scout339, 2022-06-30:
I can't say why, but on the same network with a different system (my laptop) there are no issues.

haslinghuis, 2022-06-30:
Curious what's going on but closing as incident. Please provide more info if you find out more.

closed 2022-06-30 by haslinghuis

Madflying, 2023-01-19:
I am also facing the same problem what's the solution

cm3913, 2023-01-21:
i am facing the same issue

haslinghuis, 2023-01-21:
Just retry and it works. Saw this a few times now but is intermittent