betaflight/betaflight-configurator #2994: Sliders in profiles / filters

RipperDrone, 2022-08-14:
### Describe the bug

When selecting individual profiles and setting the sliders, e.g. gyro filters, the settings do not stick. After changing from e.g. profile 1 to profile 2 and back, filter settings are lost, despite having saved them. 

### To Reproduce

like described 

### Expected behavior

Slider settings should stick to the stored values on a per profile basis. They don't.

### Configurator version

10.8.0 (release)

### Flight controller configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP API: 1.44
# config: manufacturer_id: DIAT, board_name: MAMBAF722_2022B, version: 5e174eaf, date: 2022-07-13T21:56:37Z

# start the command batch
batch start

board_name MAMBAF722_2022B
manufacturer_id DIAT

# name: Lightswitch

# feature
feature -RX_PARALLEL_PWM

# serial
serial 5 1024 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# led
led 0 12,12::C:1
led 1 13,12::C:1
led 2 14,12::C:1
led 3 15,12::C:1
led 4 12,13::C:1
led 5 13,13::C:1
led 6 14,13::C:1
led 7 15,13::C:1
led 8 12,14::C:1
led 9 13,14::C:1
led 10 14,14::C:1
led 11 15,14::C:1
led 12 12,15::C:1
led 13 13,15::C:1
led 14 14,15::C:1
led 15 15,15::C:1

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 27 0 1700 2100 0 0
aux 4 13 0 1700 2100 0 0
aux 5 30 1 900 2100 0 0
aux 6 35 2 900 1300 0 0
aux 7 49 0 1300 1700 0 0

# master
set gyro_lpf1_static_hz = 200
set gyro_lpf2_static_hz = 400
set gyro_lpf1_dyn_min_hz = 200
set gyro_lpf1_dyn_max_hz = 400
set acc_calibration = -17,-12,-21,1
set min_check = 1000
set max_check = 2000
set rssi_channel = 8
set fpv_mix_degrees = 35
set max_aux_channels = 4
set serialrx_provider = CRSF
set airmode_start_throttle_percent = 13
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set failsafe_off_delay = 25
set failsafe_throttle = 1070
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set align_board_roll = 180
set align_board_yaw = 90
set vbat_max_cell_voltage = 450
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 80
set report_cell_voltage = ON
set osd_cap_alarm = 500
set osd_vbat_pos = 2081
set osd_rssi_pos = 2088
set osd_link_quality_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2113
set osd_flymode_pos = 461
set osd_anti_gravity_pos = 385
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 2123
set osd_gps_speed_pos = 34
set osd_gps_lon_pos = 498
set osd_gps_lat_pos = 466
set osd_gps_sats_pos = 1
set osd_home_dir_pos = 100
set osd_home_dist_pos = 1
set osd_compass_bar_pos = 266
set osd_altitude_pos = 33
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 391
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 20
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED
set displayport_msp_serial = 3
set gyro_1_sensor_align = DEFAULT
set name = Lightswitch

profile 2

# profile 2
set dterm_lpf1_dyn_min_hz = 60
set dterm_lpf1_dyn_max_hz = 120
set dterm_lpf1_static_hz = 60
set dterm_lpf2_static_hz = 120
set p_pitch = 32
set i_pitch = 58
set d_pitch = 32
set f_pitch = 87
set p_roll = 31
set i_roll = 56
set d_roll = 28
set f_roll = 84
set p_yaw = 31
set i_yaw = 56
set f_yaw = 84
set d_min_roll = 21
set d_min_pitch = 23
set motor_output_limit = 80
set auto_profile_cell_count = -1
set simplified_master_multiplier = 70
set simplified_dterm_filter_multiplier = 80

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 70
set tpa_breakpoint = 1700

# end the command batch
batch end
```

haslinghuis, 2022-08-14:
It's not a bug but a feature:
[image]

RipperDrone, 2022-08-14:
> It's not a bug but a feature:
> 
> [image]

No, I knew that one for sure: Try setting the gyro filter to different settings with the slider for different profiles, with the 'Profile independent Filter Settings' switch enabled --> only the gyro filter which was set last is saved for ALL profiles. No different gyro filters per profile, as suggested...

haslinghuis, 2022-08-14:
Ah there is the misinterpretation. The `ON` button is misplaced as it's suggest toggling this function. Instead it toggles using simplified tuning sliders (ON) or manual values (OFF).

closed 2022-09-21 by github-actions[bot]