betaflight/betaflight-configurator #2997: chosen rates profile doesn't refresh on aux value change setup in adjustments tab (works on configurator 10.8)

PaulFPV, 2022-08-18:
### Describe the bug

On 10.8, when I change the rates profile (as setup in the adjustments tab), I see the selected rate profile in the pids tab. on 10.9 this doesn't work anymore.

### To Reproduce

On Betaflight Configurator 10.9.0-debug-7bc1db4_macOS

1) Setup a rates profile switching (check this video to know how to do it: https://www.youtube.com/watch?v=F6xTZkAfpR0)
2) Check that, in the adjustment tab, the aux actual value (yellow vertical line) is moving correctly
3) Go in the PID tuning tab and, again, move the switch on the radio and see if the rates profile is following correctly

= it doesn't move

4) Now try the same with a 10.8.0 configurator

= it works fine

FYI: Same behaviour on betaflight 4.3.1 and 4.4 (now that I know it isn't a betaflight bug I go back to 4.4)

### Expected behavior

As it was in 10.8, the pids tab should reflect the real selected rate profile.

### Configurator version

10.9.0-debug-7bc1db4_macOS

### Flight controller configuration

```txt
# diff all

# version
# Betaflight / STM32F405 (S405) 4.3.1 Jul 13 2022 / 03:32:11 (8d4f005) MSP API: 1.44
# config: manufacturer_id: BEFH, board_name: BETAFPVF405, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFPVF405
manufacturer_id BEFH
mcu_id 0029003d3131511939373634
signature 

# name: LQ  0:000 -130

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 B01
resource SERIAL_TX 1 NONE
resource PINIO 1 A09

# feature
feature -RX_PARALLEL_PWM
feature TELEMETRY

# serial
serial 3 1 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# led
led 0 6,8::L:2
led 1 7,8::CT:9

# color
color 1 0,0,225
color 2 41,0,255
color 3 121,0,255
color 8 10,0,255
color 9 14,0,255
color 12 129,0,255

# mode_color
mode_color 7 0 4

# aux
aux 0 0 0 1875 1925 0 0
aux 1 0 0 1625 1675 0 0
aux 2 0 0 1375 1425 0 0
aux 3 2 1 1725 1850 0 0
aux 4 13 1 1925 1975 0 0
aux 5 13 1 1725 1775 0 0
aux 6 13 1 1525 1575 0 0
aux 7 35 1 1525 1625 0 0
aux 8 40 0 1600 2025 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# rxfail
rxfail 3 s 1325

# master
set gyro_lpf1_static_hz = 312
set gyro_lpf2_static_hz = 625
set dyn_notch_count = 2
set dyn_notch_q = 350
set dyn_notch_min_hz = 120
set dyn_notch_max_hz = 500
set gyro_lpf1_dyn_min_hz = 312
set gyro_lpf1_dyn_max_hz = 625
set acc_calibration = 47,3,-22,1
set min_check = 1008
set max_check = 2000
set rc_smoothing_auto_factor = 80
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_bidir = ON
set vbat_max_cell_voltage = 422
set vbat_min_cell_voltage = 340
set force_battery_cell_count = 4
set vbat_scale = 165
set vbat_divider = 15
set ibata_scale = 804
set ibatv_scale = 600
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set gps_rescue_allow_arming_without_fix = ON
set simplified_gyro_filter_multiplier = 125
set osd_alt_alarm = 120
set osd_vbat_pos = 226
set osd_rssi_pos = 449
set osd_link_quality_pos = 384
set osd_rssi_dbm_pos = 2496
set osd_tim_2_pos = 23
set osd_flymode_pos = 2541
set osd_throttle_pos = 194
set osd_vtx_channel_pos = 21
set osd_crosshairs_pos = 271
set osd_ah_sbar_pos = 207
set osd_ah_pos = 141
set osd_current_pos = 2487
set osd_mah_drawn_pos = 2484
set osd_craft_name_pos = 2464
set osd_gps_speed_pos = 312
set osd_gps_lon_pos = 16449
set osd_gps_lat_pos = 16417
set osd_gps_sats_pos = 216
set osd_home_dir_pos = 248
set osd_home_dist_pos = 249
set osd_altitude_pos = 16662
set osd_pidrate_profile_pos = 2535
set osd_warnings_pos = 2348
set osd_avg_cell_voltage_pos = 2516
set osd_battery_usage_pos = 2508
set osd_disarmed_pos = 2316
set osd_flip_arrow_pos = 2540
set osd_rate_profile_name_pos = 16
set rate_6pos_switch = ON
set vtx_band = 0
set vtx_channel = 0
set vtx_power = 0
set vtx_freq = 0
set displayport_msp_serial = 3
set pinio_box = 40,41,42,43
set gyro_1_sensor_align = CW270FLIP
set gyro_1_align_pitch = 1800
set name = LQ  0:000 -130

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set anti_gravity_gain = 200
set iterm_relax_cutoff = 12
set pidsum_limit_yaw = 300
set p_pitch = 44
set i_pitch = 79
set d_pitch = 45
set f_pitch = 104
set p_roll = 40
set i_roll = 72
set f_roll = 95
set i_yaw = 65
set f_yaw = 95
set d_min_roll = 40
set d_min_pitch = 45
set thrust_linear = 20
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 10
set simplified_pids_mode = RP
set simplified_d_gain = 135
set simplified_pi_gain = 90
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 80
set simplified_pitch_pi_gain = 105
set simplified_dterm_filter_multiplier = 120

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 35
set thr_expo = 15
set roll_rc_rate = 21
set pitch_rc_rate = 21
set yaw_rc_rate = 18
set roll_expo = 62
set pitch_expo = 62
set yaw_expo = 62
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 65
set tpa_rate = 72
set tpa_breakpoint = 1370

rateprofile 1

# rateprofile 1
set thr_mid = 35
set thr_expo = 15
set roll_rc_rate = 13
set pitch_rc_rate = 13
set yaw_rc_rate = 11
set roll_expo = 65
set pitch_expo = 65
set yaw_expo = 65
set roll_srate = 52
set pitch_srate = 52
set yaw_srate = 44
set tpa_breakpoint = 1150

rateprofile 2

# rateprofile 2
set thr_mid = 35
set thr_expo = 20
set roll_rc_rate = 8
set pitch_rc_rate = 8
set roll_expo = 77
set pitch_expo = 77
set yaw_expo = 77
set roll_srate = 51
set pitch_srate = 51
set yaw_srate = 44
set tpa_breakpoint = 1150

rateprofile 3

# rateprofile 3
set thr_mid = 35
set thr_expo = 20
set roll_rc_rate = 4
set pitch_rc_rate = 4
set yaw_rc_rate = 4
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 38
set pitch_srate = 36
set yaw_srate = 35
set tpa_breakpoint = 1150

rateprofile 4

# rateprofile 4
set thr_mid = 35
set thr_expo = 20
set yaw_rc_rate = 6
set roll_expo = 66
set pitch_expo = 66
set yaw_expo = 66
set roll_srate = 28
set pitch_srate = 28
set yaw_srate = 24
set tpa_breakpoint = 1150
set throttle_limit_type = SCALE
set throttle_limit_percent = 70

rateprofile 5

# rateprofile 5
set thr_mid = 35
set thr_expo = 20
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_srate = 25
set pitch_srate = 25
set yaw_srate = 25
set tpa_breakpoint = 1150
set throttle_limit_type = SCALE
set throttle_limit_percent = 70

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

haslinghuis, 2022-08-18:
Thanks for the report.

joshuawang0123, 2022-08-24:
thanks foe everyone

PR #2998 (2022-09-17): Fix profile switching
Fixes: #2997
closed 2022-09-17 by blckmn