betaflight/betaflight-configurator #3007: PID Tuning Sliders Should be able to go higher.

malinhiles, 2022-08-24:
### Is your feature request related to a problem? Please describe

The sliders don't move far enough to properly tune small multirotors. My small drones (3" propeller size and below) need higher PIDs than what is possible to achieve with the sliders.

### Describe the solution you'd like

Via Expert Mode, add the ability to adjust the endpoints of the tuning sliders to provide more latitude when tuning different size aircraft.

### Describe alternatives you've considered

Typing in pid values manually, which disables sliders entirely.

malinhiles, 2022-09-24:
Any thoughts on this?

CptLuki, 2022-09-24:
You have MM slider and dedicated PI and D sliders. You can max both if you need lot of gain. Honestly in typical builds you are never going to max both so not sure why your 3inch needs that much gain. Anyway if you need more than that you can still go in and type values manually. Just use calc to keep ratios.

closed 2022-11-02 by github-actions[bot]