betaflight/betaflight-configurator #3009: Incorrect permissions set in DEB package on recent nightlies

howels, 2022-08-28:
### Describe the bug

```
-rwx------ 1 root root 226K Aug 23 23:08 /opt/betaflight/betaflight-configurator/betaflight-configurator
```

BetaFlight configurator should have `-rwxr-xr-x` permissions but recent nightlies show `-rwx------` which prevents regular users executing configurator.

The file `/opt/betaflight/betaflight-configurator/chrome_crashpad_handler` also lacks incorrect execute permissions and the entire betaflight-configurator directory is only readable by root. 

Present in `betaflight-configurator_10.9.0-debug-ea02d9a_amd64.deb`

### To Reproduce

Install deb package, e.g. `betaflight-configurator_10.9.0-debug-ea02d9a_amd64.deb`

### Expected behavior

Unix permissions should be present as they were in prior releases including 10.8

### Configurator version

10.9.0-debug-ea02d9a

howels, 2022-09-28:
this is still present in recent releases (10.9.0-debug-db99a98).  Please can someone look at this?

howels, 2022-09-28:
In fact the deb is further broken now, missing the libnw.so library:

```
/opt/betaflight/betaflight-configurator/betaflight-configurator: error while loading shared libraries: libnw.so: cannot open shared object file: No such file or directory
```

howels, 2022-09-28:
Suggested fix:

```
$ chmod +xr /opt/betaflight/betaflight-configurator/chrome_crashpad_handler 
$ chmod +xr /opt/betaflight/betaflight-configurator/betaflight-configurator
$ chmod -R +Xr /opt/betaflight/betaflight-configurator/
```

haslinghuis, 2022-09-28:
The last line should be sufficient?

PR #3028 (2022-10-04): Fix permissions in debian package
Fixes: #3009
closed 2022-10-04 by haslinghuis