betaflight/betaflight-configurator #3012: Omnibus F4SD connection problem with 10.8

kondzio-p, 2022-08-31:
### Describe the bug

On 10.7 I can normally connect with my Omnibus F4SD. 
I then try to hop on 10.8 and it just doesn't work. 
Same with 10.9-debug
[image]
[image]
[image]

### To Reproduce

Plug in a Omnibus F4SD fc and run betaflight 10.8.

### Expected behavior

I expect to connect with my FC normally.

### Configurator version

10.8, 10.9-debug

McGiverGim, 2022-09-02:
I only can say that I can connect with Betaflight Configurator 10.8.0 to my OmnibusF4SD without problem.

Do you see some port in the combo of Manual Selection or is empty?

haslinghuis, 2022-09-02:
What does device manager ports and usb show?

kondzio-p, 2022-09-03:
The combo of manual selection is empty, I downloaded all the drivers, ran the fc through the driver fixer. I literally can’t find a post or video from someone who has the same problem.

Asizon, 2022-09-03:
@kondzio-p this is clearly drivers issue, try reverting to the previous driver or using zadig

closed 2022-10-11 by github-actions[bot]