betaflight/betaflight-configurator #3014: Bluetooth and Wifi serial comunication

MarcinKnD, 2022-09-01:
### Is your feature request related to a problem? Please describe

A great idea is to extend BetaFlight Configurator with connectivity via BT and WiFi SERIAL. Modules such as ESP01s are small, light and with ESPlink software installed, they act as a WIFI bridge to the serial. The connection is very simple, only no support in the configurator. While Virtual Com can be helped on a PC, it is not possible on an Android phone. Make it possible to connect via SERIAL WIFI and BT

### Describe the solution you'd like

Support for SERIAL WIFI and SERIAL Bluetooth connection

### Describe alternatives you've considered

no more usb cable

wouldn't that be great

Hawxxer, 2022-09-02:
Related to this https://github.com/betaflight/betaflight-configurator/issues/466 
Looks like normal bluetooth already works.

haslinghuis, 2022-09-05:
Duplicate of #466

closed 2022-09-05 by haslinghuis