betaflight/betaflight-configurator #3017: Request dummy PIDs and active sliders in 'Virtual Mode'....

SupaflyFPV, 2022-09-03:
### Is your feature request related to a problem? Please describe

This might be a bit specialist but I often find myself offering support and needing to recommend slider changes, but I don't know them off my head but I could look up in the configurator virtual mode...however the PID window is empty...

### Describe the solution you'd like

Would it be possible to populate the pids with the default numbers - or a preset?

### Describe alternatives you've considered

something alternative

### Other information

no screenshot

Prajwal-Koirala, 2023-12-27:
**Here's a summary of the feature request and potential solutions:**

**Problem:**

- The PID window in the configurator's virtual mode is empty, making it difficult to recommend slider changes without knowing default values or presets.

**Requested Solution:**

- Populate the PID window with either:
    - Default PID values
    - Optional presets for common configurations

**Alternatives Considered:**

- The user mentions "something alternative" but doesn't provide specifics.

**Additional Information:**

- No screenshot is provided.

**Potential Solutions:**

1. **Display Default PID Values:**
   - Automatically populate the PID window with default values upon entering virtual mode.
   - Provide an option to toggle this feature on/off based on user preference.

2. **Implement PID Presets:**
   - Create a selection of predefined PID presets for common configurations.
   - Allow users to load these presets into the PID window, saving time and effort.

3. **Combine Both Solutions:**
   - Offer both default PID values and preset options for maximum flexibility.

**Next Steps:**

- Gather more information about specific use cases and user preferences to determine the most suitable solution.
- Evaluate technical feasibility and implementation effort for each option.
- Consider potential impact on configurator performance and user experience.
- Prioritize the solution that best aligns with user needs and aligns with development resources.

haslinghuis, 2024-05-06:
Rejecting request as sliders require firmware interaction after firmware 4.3 it would be hard to implement and maintain.

closed 2026-01-14 by haslinghuis