betaflight/betaflight-configurator #3019: Can't flash  411 AIO board using BFC 10.8.0

porlock, 2022-09-05:
### Describe the bug

Hey, I tried to flash BF 4.2.11 Tinyhawk FS II board with target MatekF411RX and Moblite 7 diamond AIO with target CRAZYBEEF4FR.  Both stuck on "erasing" part, nothing more happens.  On BF configurator 10.7.2 everything worked at first try.

### To Reproduce

Try flash BF 4.2.11 using BFC 10.8.0 on
- Tinyhawk FS II board with target MatekF411RX 
- Moblite 7 diamond AIO

### Expected behavior

boards flash successfully 

### Configurator version

10.8.0

duanfpv, 2022-10-04:
how to download

closed 2022-10-23 by Asizon

porlock, 2022-10-23:
why its closed