betaflight/betaflight-configurator #3021: Cannot set pidsum through cli

ApolloBian, 2022-09-09:
### Describe the bug

I have set pidsum_limit & pidsum_limit_yaw both to 1000 through cli:
[image]

However, in blackbox explorer, the traces show that pidsum is only at 500(50%):
[image]
the info panel also shows that pidsum is not set:
[image]

### To Reproduce

Betaflight configurator windows 10.8.0 (5ec9465)
FC: HGLRCF722(STM32F7X2)

### Expected behavior

pidsum_limit should be set to 1000

### Configurator version

Betaflight configurator windows 10.8.0 (5ec9465)

### Flight controller configuration

```txt
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.1 Jul 13 2022 / 03:36:10 (8d4f005) MSP API: 1.44
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: 907ed680, date: 2022-06-29T01:35:35Z

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# name: TP3

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature OSD

# serial
serial 2 8192 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 3 1700 2100 0 0
aux 2 35 2 1700 2100 0 0
aux 3 39 1 1400 2100 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 25 100 200

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 0
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 1000
set acc_calibration = 85,-25,-4057,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1002
set max_check = 1990
set rc_smoothing_auto_factor = 52
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_idle_value = 400
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set yaw_deadband = 5
set simplified_gyro_filter_multiplier = 200
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2444
set osd_tim_2_pos = 2454
set osd_remaining_time_estimate_pos = 375
set osd_g_force_pos = 2094
set osd_current_pos = 2080
set osd_mah_drawn_pos = 2103
set osd_craft_name_pos = 2433
set osd_gps_lat_pos = 49320
set osd_battery_usage_pos = 34904
set osd_flip_arrow_pos = 2286
set osd_log_status_pos = 2169
set osd_camera_frame_pos = 49152
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 2
set vtx_power = 1
set vtx_freq = 5760
set gyro_1_sensor_align = CW180
set gyro_1_align_pitch = 0
set rpm_filter_harmonics = 1
set rpm_filter_q = 750
set rpm_filter_min_hz = 125
set name = TP3

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 82
set dterm_lpf1_dyn_max_hz = 165
set dterm_lpf1_static_hz = 165
set dterm_lpf2_static_hz = 165
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 51
set i_pitch = 92
set d_pitch = 50
set f_pitch = 137
set p_roll = 49
set i_roll = 88
set d_roll = 44
set f_roll = 132
set p_yaw = 49
set i_yaw = 88
set f_yaw = 132
set d_min_roll = 33
set d_min_pitch = 37
set auto_profile_cell_count = 4
set dyn_idle_min_rpm = 34
set simplified_master_multiplier = 110
set simplified_dterm_filter = OFF
set simplified_dterm_filter_multiplier = 110

rateprofile 0

# rateprofile 0
set rates_type = KISS
set roll_rc_rate = 198
set pitch_rc_rate = 198
set yaw_rc_rate = 165
set roll_expo = 21
set pitch_expo = 21
set yaw_expo = 21
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 64
set tpa_rate = 70
set throttle_limit_percent = 85

# end the command batch
batch end

#
```

### Add any other context about the problem that you think might be relevant here

Is there any added hidden switch that toggles pidsum_limit? I used to set pidsum_limit in cli and it worked.

ApolloBian, 2022-09-09:
Also a blackbox log after I have set the pidsum:
[btfl_004.txt](https://github.com/betaflight/betaflight-configurator/files/9535459/btfl_004.txt)

haslinghuis, 2022-09-09:
I see the pidsum has been set for profile 0 in firmware:
```
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
```