betaflight/betaflight-configurator #3022: cant figure out why its not workinf

Steve-fpv92, 2022-09-16:
### Describe the bug

says i need to update but i have the latest and it keeps downloading the same thibng can someone help im bad with online looking due to personal stuff. stoodley603@gmail.com

### To Reproduce

just uninstall then install but wont fix

### Expected behavior

help

### Configurator version

10.8.0

### Flight controller configuration

```txt
dji r1
```

### Add any other context about the problem that you think might be relevant here

please call or email stoodley603@gmail.com win 11 lenovo laptop

Steve-fpv92, 2022-09-16:
anyone who knows beta like a pro hmu plz lmk

haslinghuis, 2022-09-16:
Duplicate of https://github.com/betaflight/betaflight-configurator/issues/2986

closed 2022-09-16 by haslinghuis