betaflight/betaflight-configurator #3040: Installer does not allow installation on Windows 11 on ARM

Wikiwix, 2022-10-10:
### Describe the bug

When I try to install Betaflight Configurator 10.8.0 on an ARM64 Windows 11 laptop it closes with an error message explaining that the installer only supports amd64. Fact is Windows11 ARM notebooks (Windows 10 ARM notbooks cannot though) [can emulate amd64](https://blogs.windows.com/windows-insider/2020/12/10/introducing-x64-emulation-in-preview-for-windows-10-on-arm-pcs-to-the-windows-insider-program/).

The portable version of the configurator even works as expected.

### To Reproduce

Run `winget install Betaflight.Betaflight-Configurator` on an ARM64 laptop.

### Expected behavior

The installer successfully installs the configurator

### Configurator version

10.8.0

### Add any other context about the problem that you think might be relevant here

I am willing to test a new installer, because I guess no ARM64 Windows Laptop might be available for you

McGiverGim, 2022-10-10:
I didn't know the new arm devices can emulate win64. I will look into the installer to see if we have some way of detect it.

McGiverGim, 2022-10-10:
It seems that is a feature that is not provided at this moment, but is in progress. In this thread there is a workaround adding code to the installer https://stackoverflow.com/questions/73707415/can-inno-setup-detect-windows11-on-arm64-hardware-which-can-emulate-x64

daylenpwns, 2022-11-12:
how would i do it with a raspberry pi os? i tried downloading but that didnt work for me

Prajwal-Koirala, 2023-12-27:
As the question you have linked shows, you can call [`GetMachineTypeAttributes` WinAPI function](https://docs.microsoft.com/en-us/windows/win32/api/processthreadsapi/nf-processthreadsapi-getmachinetypeattributes) on Windows 11 to determine x64 support. On older versions of Windows, use [`ProcessorArchitecture` Inno Setup support function](https://jrsoftware.org/ishelp/index.php?topic=isxfunc_processorarchitecture)

Something like this should do (but I do not have ARM64 machine to test this on). 

```pascal
[Code]

const
  IMAGE_FILE_MACHINE_AMD64 = $8664;

function GetMachineTypeAttributes(
    Machine: Word; var MachineTypeAttributes: Integer): HRESULT;
  external 'GetMachineTypeAttributes@Kernel32.dll stdcall delayload';

<event('InitializeSetup')>
function InitializeSetupCheckArchitecture(): Boolean;
var
  MachineTypeAttributes: Integer;
  Arch: TSetupProcessorArchitecture;
begin
  if IsWindows11OrNewer then
  begin
    OleCheck(
      GetMachineTypeAttributes(IMAGE_FILE_MACHINE_AMD64, MachineTypeAttributes));
    if MachineTypeAttributes <> 0 then
    begin
      Log('Windows 11 or newer, with x64 support');
      Result := True;
    end
      else
    begin
      Log('Windows 11 or newer, without x64 support');
      Result := False;
    end;
  end
    else
  begin
    Arch := ProcessorArchitecture;
    if Arch = paX64 then
    begin
      Log('Windows 10 or older, on x64 architecture');
      Result := True;
    end
      else
    begin
      Log('Windows 10 or older, not on x64 architecture');
      Result := False;
    end;
  end;

  if not Result then
  begin
    MsgBox('This product can be installed on system with x64 support only.',
      mbError, MB_OK);
  end;
end;
```

The code completely replaces the `ArchitecturesAllowed` directive (it assumes it is not set, though setting it to `x64 arm64` should do no harm).

The `IsWindows11OrNewer` comes from https://stackoverflow.com/q/5849917/850848.

haslinghuis, 2024-05-05:
We are moving away from NWjs embracing PWA.

closed 2024-05-05 by haslinghuis