betaflight/betaflight-configurator #3048: Betaflight Configurator 10.8.0 APK : "Obsolete" message

Canaill51, 2022-10-14:
### Describe the bug

When opening Betaflight Configurator Android app, it always reports that it is obsolete. Even if I clean its cache on my Android device, and reinstall the app, still reporting this message.
[image]

### To Reproduce

1. Install the latest Android APK from https://github.com/betaflight/betaflight-configurator/releases/download/10.8.0/betaflight-configurator_10.8.0-debug-5ec9465_android.apk
2. Run BF10.8.0 GUI app.

### Expected behavior

BF Configurator should open without any out-of-date message reporting, because it is up to date.

### Configurator version

10.8.0-debug-5ec9465

deadpool772, 2022-11-06:
Just ran into this trying to install betaflight on my phone for the first time ever. I'm new to the FPV hobby this year, hopefully there's a fix to this soon

haslinghuis, 2022-12-07:
It will be fixed in next version.

closed 2023-01-16 by github-actions[bot]

123GoqkadMe, 2025-01-27:
- [ ] betaflight configurator nightly 
- [ ] download