betaflight/betaflight-configurator #3049: Layout misaligned in android configurator nightlies

PSi86, 2022-10-14:
### Describe the bug

Hi Team,
In the 10.9 nightly builds the androids configurator layout is not aligned correctly.

On the right side of the screen the main frame of the app does not use the display to the edge. And the menu on the left side does not show any text - just the symbols.

[image]

### To Reproduce

Install 10.9 android configurator nightly, open it on an Android 12 device and observe the layout. I am using a Samsung S21.

### Expected behavior

Main application frame should use the display to the right edge and the menu bar should show the labels.

### Configurator version

10.9.0-debug-b5c2216

ASDosjani, 2022-10-17:
Can you check which is the first version where it starts happening?

McGiverGim, 2022-10-17:
I think it's when we moved from css to less. I suppose some rule was converted in a wrong way.

ASDosjani, 2022-10-17:
@McGiverGim you are right. I checked and it happens after #2929. v20220623.130

PSi86, 2022-10-17:
Thank you for checking and finding the culprit. If you need me for testing, please ping me here.

PSi86, 2022-10-17:
Wrong button..

ASDosjani, 2022-10-22:
[image]
This gray line on the right is also related to #2929.

closed 2022-10-25 by blckmn