betaflight/betaflight-configurator #3050: ELRS device dissapears after a second

ASDosjani, 2022-10-16:
### Describe the bug

After the configurator starts it discovers the elrs 3.0 device but it dissapears after a second. Based on my testings this bug happens after in the port_handler.js the `TCP_CHECK_INTERVAL` expires and the tcpCheck() method called in line 107.
[image]

If I power my drone after the configurator loaded than it connects automatically but then it disconnect.
[image]

I can connect to my drone manually via wifi.

### To Reproduce

Plug the drone in and check the configurator.

### Expected behavior

Not dissapearing.

### Configurator version

10.9

freasy, 2022-10-24:
Hello :)

yes i can take a look into that.

freasy, 2022-11-26:
> @freasy this bug is still relevant.

I will look into this today

ASDosjani, 2022-11-26:
> 

I found the bug and made a PR, check #3095

ASDosjani, 2022-11-27:
@freasy do you have discord? I would glad if you could join to the Betaflight server.
I still have some problem on Android side.

PR #3095 (2022-12-02): Fix elrs connection
The previous .get() method was broken, `self.mdnsBrowser.init` was pointless and the `remove()` function does not delete the services properly.
This PR hopefully fixes these.
On android it doesn't remove offline devices but disables and pings them while they become online again. I also copied the connect button click code from serial backend only to be unified.

Thanks to @haslinghuis the whole mdns code went to a different class thus keeping the code clean.

Fixes #3050
Merges: #3094
closed 2022-12-02 by haslinghuis