betaflight/betaflight-configurator #3064: PID Profile Names

malinhiles, 2022-11-01:
### Is your feature request related to a problem? Please describe

I carry a range of different payloads on my cinelifter and 5" aircrafts at different altitudes. I establish different PID tunes for the different payload types that I carry. Depending on what I payload I need to fly, I select different PID profiles to match my situation.

It would be really helpful if each PID Profile in the configurator could be nick-named so that you wouldn't have to remember or make external note of what each PID profile corresponded to. 

In addition to that, if there was any more room to store more PID profiles, like maybe 6 instead of 3, that'd be great too.

### Describe the solution you'd like

Allow pid profiles to be given alphanumeric nicknames in the configurator.

### Describe alternatives you've considered

A note in my phone to remember what PID profile is for what payload profile

qvasic, 2022-11-02:
Actually, you can assign names to PID profiles by setting profile_name variable via CLI and then see your active profile name via OSD element "Profile: PID profile name".

Yet this could be improved:
1, Changing profiel name using configurator itself.
2. Showing profile name while scrolling through them in OSD menu.
3. Increasing PID profiles count.

malinhiles, 2022-11-03:
@qvasic Ahh, I see. I didn't know that the name could be changed in CLI, for use in the OSD. Even still, a different and improved implementation would better address the issue. 

When tuning PIDs and changing PID profiles in the configurator's "PID Tuning" tab, the profiles are still called "PID Profile 1",  "PID Profile 2", etc. which is hard to keep track of across multiple types of aircraft, carrying different payloads, for different conditions. If I could rename the PID profiles (and maybe even the rate profiles too) in the configurator, that would mean that i no longer have to remember what tune corresponds to each PID profile number. This I feel would directly solve the issue. 

> Yet this could be improved:
> 1, Changing profiel name using configurator itself.
> 2. Showing profile name while scrolling through them in OSD menu.
> 3. Increasing PID profiles count.

Changes 2 and 3 that you mentioned are great too.

qvasic, 2022-11-03:
Rate profile names can be changed/shown too, CLI variable name rateprofile_name, corresponding OSD element.
I'll see what I can do about improving other things too.

qvasic, 2022-11-04:
Also, apparently in the Betaflight master PID profile count is already 4. So it will probably be so in the next release.

closed 2022-12-17 by KarateBrot

TomTom4221, 2023-05-01:
Profile names (PID & Rate) are great. Much easier to remember what was applied. 
1) However the 8 character limit makes my descriptions very cryptic. Any chance for e.g. a 16 character field?
2) Just cosmetics: I would rename in the PID tuning TAB the 'Profile' selector (which is PID profile) into 'PID Profile'. 
THANKS!

PR #3150 (2022-12-17): Fields for PID and rate profile names.
Fixes #3064

Some pilots have different PID profiles for different payloads/conditions they fly(lighter/bigger battery, action camera/no camera, various elevations), different rate profiles for different styles of flying (cine, race, freestyle).
Having readable profile names would significantly simplify selecting the proper profile, as opposed to memorizing profile numbers - just imagine having N quads with M profiles each.

[image]

[image]