betaflight/betaflight-configurator #3072: Increase scale for gyro readings in Motor and Sensor Tab

SupaflyFPV, 2022-11-06:
### Is your feature request related to a problem? Please describe

Currently there are a lot of gyro issues coming up due to QC issues. I've just encountered one with a BMI gyro with high baseline noise. 

In the motor tab there is a chart showing the active gyro outputs and the scale goes up to 10. This makes it a little easier to observe the noise floor. On the 'Sensors' tab the scale only goes up (or down) to 50 which means the noise floor can barely be seen.

I am asking if we could get further scaling on both to make it easier to observe the noise floor and look for issues without having to fly and take a black box log.

### Describe the solution you'd like

Scale on both readouts to perhaps 5 or even bigger....

### Describe alternatives you've considered

Logging is the only other alternative. However if we can get a proper scale then we can establish levels for normal and problematic noise levels without the need to fly/log when doing support.

SupaflyFPV, 2022-11-06:
[image]

haslinghuis, 2022-11-06:
Let me know which scales are usefull

SupaflyFPV, 2022-11-06:
> Let me know which scales are usefull

could you add 5 and 2 and 1? I'm not sure how the scaling works...the image above is the biggest it can currently go set at 10...

haslinghuis, 2022-11-06:
Please test 3073, added a lot more

haslinghuis, 2022-11-06:
[image]

SupaflyFPV, 2022-11-06:
awesome will test now

SupaflyFPV, 2022-11-06:
thats brilliant thanks - really useful now

If I was greedy I'd ask if the constant peak values per axis can exist in the sensors tab like it does in the motors?

SupaflyFPV, 2022-11-06:
for anyone interested you can straight away see the noisy BMI gyro I have here compared to a known working MPU6000
[image]
[image]

closed 2022-11-10 by haslinghuis