betaflight/betaflight-configurator #3075: After successfully modifying the power of VTX in the 4.3.0/4.3.1 firmware, it will not take effect after restarting.

yanwenjun, 2022-11-09:
### Describe the bug

After successfully modifying the power of VTX in the 4.3.0/4.3.1 firmware, it will not take effect after restarting.
But the same operation does not have this problem in version 4.2.11.
vtx model：SpeedyBee TX800
protocol：IRC Tramp

### To Reproduce

VTX power list:
[image]
1.Set the power of the vtx to 800 mW, 
[image]
2.Click Save, and the orange light of SpeedyBee-TX800 will light up (the product manual indicates that the orange light is Constantly ON, indicating that the power of 800 mW has been turned on).
3.Finally restart, the orange light of VTX will light off.
4.At this time, go to the Ports tab to turn off the VTX serial port so that the VTX does not accept the configuration from the flight controller. reboot the flight controller, the orange light of the VTX will Constantly ON, indicating that an incorrect power is set for the vtx when the flight controller restarts. this problem.

Doing the same in BF-4.2.11 version, this problem doesn't happen.

### Expected behavior

After restarting the plane was able to keep the 800mW vtx setting instead of reverting to 25mW.

### Configurator version

10.8.0

### Flight controller configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.11 Nov  9 2021 / 20:29:32 (948ba6339) MSP API: 1.43
# config: manufacturer_id: SPBE, board_name: SPEEDYBEEF7MINI, version: edcd244b, date: 2022-06-29T02:24:46Z

# start the command batch
batch start

board_name SPEEDYBEEF7MINI
manufacturer_id SPBE

# feature
feature -RX_PARALLEL_PWM

# serial
serial 2 8192 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 200 400 600
vtxtable powerlabels 25 200 400 800

# master
set vtx_power = 4

profile 0

rateprofile 0

# end the command batch
batch end
```

McGiverGim, 2022-11-10:
I'm not too sure if this is a Configurator problem or a firmware one. What I can see is that `vtx_power = 4` seems wrong, you only have 4 power values so `vtx_power` must be between 0 and 3.
If you change the `vtx_power` to 3 in CLI it works?

yanwenjun, 2022-11-11:
Sorry, I copied it wrong, the correct power list looks like this.
But the problem he presents is the same.

[image]

yanwenjun, 2022-11-11:
This also happens when I set vtx_power to a value between 1 and 3.

yanwenjun, 2022-11-17:
> I'm not too sure if this is a Configurator problem or a firmware one. What I can see is that `vtx_power = 4` seems wrong, you only have 4 power values so `vtx_power` must be between 0 and 3. If you change the `vtx_power` to 3 in CLI it works?
[image]
A value of 0 acts as the None option
This also happens when I set vtx_power to a value between 1 and 3.