betaflight/betaflight-configurator #3100: `master` - OSD tab - HD grid scale/z-level issue

nerdCopter, 2022-11-29:
### Describe the bug

* when configurator window is rather narrow, and `HD` is selected, the OSD grid pattern is scaled wrongly and z-level is above the options.
* maybe in reference to https://github.com/betaflight/betaflight-configurator/pull/3071
* tested/found on master `8aced3b3`

### To Reproduce

* On size the window about 1400 pixels wide.
* select HD
* try to select any other option in the same verical level as the OSD-background-image.

### Expected behavior

* options should be selectable

### Configurator version

`8aced3b3`

KarateBrot, 2022-11-29:
The grid is a "pre-baked" image afaik. Can we maybe just draw the gridlines dynamically "on the fly" using d3.js or something similar?

McGiverGim, 2022-11-29:
It was changed to a dinamically one in the HD PR. What it seems to me is that the grid is ok, but the window is not. It does not have the correct proportion.

haslinghuis, 2022-11-29:
Seems working with above fix. Might not be the right solution but it works now on smaller screens.

PR #3101 (2022-12-05): Fix OSD preview
Fixes: #3100
closed 2022-12-05 by haslinghuis