betaflight/betaflight-configurator #3131: Upgrade the artifacts comment creator

McGiverGim, 2022-12-09:
### Describe the bug

The arifacts comment creator action is deprecated:
>Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: tonyhallett/artifacts-url-comments@v1.1.0

This issue is only opened to not forgot it. We are at the latest version and there have not too much options for this. But at some moment (estimated summer 2023) we will need to upgrade the version or move to another action.

### To Reproduce

Look at the github actions log.

### Expected behavior

Not to have warnings in the Actions log.

### Configurator version

-

### Flight controller configuration

```txt
-
```

### Add any other context about the problem that you think might be relevant here

-

McGiverGim, 2022-12-09:
Only as idea, if we can't find any better action, we can do it "manually", using the REST API of Github to list the artifacts, and processing the JSON and adding the comment.
Here info about the API: https://docs.github.com/en/rest/actions/artifacts?apiVersion=2022-11-28

haslinghuis, 2023-06-27:
Was this not resolved in #2813

closed 2023-06-27 by haslinghuis