betaflight/betaflight-configurator #3135: Incorrect permissions set in RPM package in 10.9-RC1

alteman, 2022-12-11:
### Describe the bug

 #3009 still an issue for RPMs
@howels

-rwx------ 1 root root 226K Aug 23 23:08 /opt/betaflight/betaflight-configurator/betaflight-configurator

BetaFlight configurator should have -rwxr-xr-x permissions but recent nightlies show -rwx------ which prevents regular users executing configurator.

The file /opt/betaflight/betaflight-configurator/chrome_crashpad_handler also lacks incorrect execute permissions and the entire betaflight-configurator directory is only readable by root.

Present in betaflight-configurator_10.9.0-RC1

### To Reproduce

Install RPM package, e.g. betaflight-configurator-10.9.0-RC1.x86_64.rpm 

### Expected behavior

Unix permissions should be present as they were in prior releases including 10.8

### Configurator version

10.9.0 (7df8d62)

haslinghuis, 2022-12-11:
@alteman please try:
```
$ chmod +xr /opt/betaflight/betaflight-configurator/chrome_crashpad_handler 
$ chmod +xr /opt/betaflight/betaflight-configurator/betaflight-configurator
$ chmod -R +Xr /opt/betaflight/betaflight-configurator/
```

howels, 2022-12-11:
I think I can fix this.

howels, 2022-12-12:
@haslinghuis beat me to this one. 🙂

closed 2022-12-12 by haslinghuis