betaflight/betaflight-configurator #3141: BT SPP device NOT found

lida2003, 2022-12-13:
### Describe the bug

I tried mobile app with latest version [Betaflight Configurator 10.9.0-RC1](https://github.com/betaflight/betaflight-configurator/releases/tag/10.9.0-RC1)

I found this:
1) My BT SPP device is OK iNav Configurator, and works fine for settings/CLI commands
2) My BT SPP device is NOT recognized by BF PC configurator
3) My BT SPP device is NOT recognized by BF mobile app
4) Android BT UART tool recognize the BT SPP device, and works fine if using MSP protocols.

Is there any thing wrong?  Is it a bug?

[image]

[image]

[image]

[image]

### To Reproduce

It can be 100% sure on my device.

### Expected behavior

BT SPP device can be recognized as iNav do, on mobile app and PC configurator.

### Configurator version

10.8.0 10.9.0-RC1

haslinghuis, 2022-12-13:
Please try enabling `showAllSerialDevices` in `options` tab:

[image]

If this works we like to know the `vendorId` and `productId` so we can add it to the serial device list.
`

lida2003, 2022-12-14:
Yes, I did enable showAllSerialDevices option, BUT no seiral port popup.

I also checked ESP32 demo code, there is NO vendor id and product id also (I have posted a message on ESP forum ask about this).

Is there any thing I can try? @haslinghuis 

[image]

haslinghuis, 2022-12-14:
Is this specific to Android or also to PC (Windows, Linux, Mac)?

Using `showAllSerialDevices` we disable the filter on allowed serialDevices

lida2003, 2022-12-16:
I saw the COM8 for SnapAirUnit (which is BT SPP device) on windows tool.

But it didn't work, no MSP message from the configurator. @haslinghuis  is it vendor id and product id issue?

[image]

lida2003, 2022-12-16:
I noticed that both iNav Configurator (PC) & BT SPP mobile test app send msp messages to FC, **BUT BF Configurator (PC) will NOT send msp message, it just connect the port only.**

-----------

- iNav Configurator (PC)  -- BT SPP UART -- FC
```
I (9349) spp: Own address:[24:6f:28:b3:f3:1e]
SnapAirUnit> W (24269) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (24529) BT_APPL: new conn_srvc id:26, app_id:255
I (25399) msp: handle_msp_protocol 9 bytes cmd 0x0001-1 flag 0 size 0
I (25399) msp: 24 58 3c 00 01 00 00 00 45
I (25409) msp: ttl_handle_msp-enter 12 bytes
I (25419) msp: 24 58 3e 00 01 00 03 00 00 02 04 73
... bla... bla...
W (35689) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x41 current:0x0
W (35699) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x41
E (35699) BT_BTC: btc_spp_write unable to find RFCOMM slot!
W (38799) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x13
W (137479) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
```

- BT SPP mobile test app  -- BT SPP UART -- FC  // hex input mannually

```
W (191299) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (191749) BT_APPL: new conn_srvc id:26, app_id:255
I (194479) msp: handle_msp_protocol 9 bytes cmd 0x0003-3 flag 0 size 0
I (194489) msp: 24 58 3c 00 03 00 00 00 cf
I (194499) msp: ttl_handle_msp-enter 12 bytes
I (194499) msp: 24 58 3e 00 03 00 03 00 05 00 00 96
W (201699) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
W (201729) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0
W (239249) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0
I (239319) msp: handle_msp_protocol 9 bytes cmd 0x0001-1 flag 0 size 0
I (239329) msp: 24 58 3c 00 01 00 00 00 45
I (239339) msp: ttl_handle_msp-enter 12 bytes
I (239339) msp: 24 58 3e 00 01 00 03 00 00 02 04 73
W (246539) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
W (246599) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0
W (429749) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:1 state 2 closed: Closed (res: 19)
W (431249) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x0
W (431269) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x41 current:0x0
W (431279) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x41
W (435279) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (435439) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16
```

- BF Configurator (PC)  -- BT SPP UART -- FC

```
W (137749) BT_APPL: new conn_srvc id:26, app_id:255
W (147789) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:1 state 2 closed: Closed (res: 19)
W (147819) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x41 current:0x0
W (147819) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x41
W (150919) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13
W (191239) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
```

haslinghuis, 2022-12-17:
Yes looked at the implementation we don't have BLE implemented

lida2003, 2022-12-19:
Related topics: 

a) Bluetooth and Wifi serial comunication #3014  //  Sep 2, 2022
b)  [Feature request] BLE (HM10/HM11) support #466 // Mar 12, 2017

lida2003, 2023-11-07:
[image]

```
Device BTHENUM\Dev_246F28B3F31E\7&297c82f3&0&BluetoothDevice_246F28B3F31E was configured.

Driver Name: bth.inf
Class Guid: {e0cbf06c-cd8b-4647-bb8a-263b43f0f974}
Driver Date: 06/21/2006
Driver Version: 10.0.22621.1928
Driver Provider: Microsoft
Driver Section: BthGenericDevice.NT
Driver Rank: 0xFF2000
Matching Device Id: BTHENUM\GENERIC_DEVICE
Outranked Drivers: 
Device Updated: false
Parent Device: BTH\MS_BTHBRB\6&e3528ed&0&1
```

lida2003, 2023-12-28:
> This issue is about a Bluetooth Serial Port (SPP) device not being recognized by Betaflight Configurator. The user is able to connect the device with other applications, but Betaflight Configurator does not recognize it.
> 
> The developer has asked the user to set their output to English and to clarify whether the issue is specific to Android or also occurs on PC.

It works well with [inav/bf configurator on linux](https://blog.csdn.net/lida2003/article/details/127901773)

haslinghuis, 2024-05-09:
NWjs and chrome API does not support BLE.

apk015, 2024-06-11:
**Subject: Help Needed for Wireless Bluetooth Connection with Betaflight/INAV**

Hi everyone,

I need help setting up a wireless Bluetooth connection for Betaflight and INAV using a SpeedyBee F405 flight controller with built-in Bluetooth. I've got the web Betaflight configurator working and can connect, but I need advice on maintaining a stable connection and integrating this with Betaflight and INAV configurators. Any tips or resources would be greatly appreciated!

Thanks,
SURAJ ( FLYSPARK )

haslinghuis, 2025-03-21:
Outdated.

closed 2025-03-21 by haslinghuis