betaflight/betaflight-configurator #3160: Betaflight Configurator 10.9.0 RC2: Error when importing backup from 4.4.0 RC1

Canaill51, 2022-12-22:
### Describe the bug

I upgraded a flight controller to 4.4.0 RC1 within GUI 10.9.0 RC2.
I set it from scratch and made a backup from Preset menu -> Save backup.
When flashing another FC with same target (a spared one), I got some error messages when loading this previous backup done.

### To Reproduce

- Windows 11 64bits.
- BF GUI 10.9.0 RC2 (bc6307a)

I created this video to explain steps and how I produce these error messages:
https://drive.google.com/file/d/1fEG4w7sS3f52gYi7dWHGHbMAcmgrTxRe/view?usp=sharing

Also, attached here a diff that occurs the  issue.

### Expected behavior

When loading backup from Preset menu, when same rev of BF, there should not be these error messages, making confusion to users: What is wrong on my backup?

### Configurator version

10.9.0 RC2 (bc6307a)

### Flight controller configuration

```txt
defaults nosave

# version
# Betaflight / STM32F405 (S405) 4.4.0 Dec 19 2022 / 16:20:37 (103f238ee) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 004100185553500920323156
signature 

# name: APEX5

# feature
feature TELEMETRY

# serial
serial 2 64 115200 57600 0 115200
serial 5 131073 115200 57600 0 115200

# beeper
beeper -ON_USB

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 13 3 1300 2100 0 0
aux 3 35 3 1700 2100 0 0
aux 4 36 1 1700 2100 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 850
set dyn_notch_count = 2
set dyn_notch_q = 400
set gyro_lpf1_dyn_min_hz = 425
set gyro_lpf1_dyn_max_hz = 850
set acc_calibration = 62,-9,96,1
set rc_smoothing_auto_factor = 140
set rc_smoothing_setpoint_cutoff = 25
set rc_smoothing_feedforward_cutoff = 25
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 3,2,1,0,4,5,6,7
set failsafe_throttle = 1070
set failsafe_procedure = AUTO-LAND
set align_board_yaw = 180
set vbat_display_lpf_period = 10
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 170
set report_cell_voltage = ON
set osd_vbat_pos = 3748
set osd_rssi_pos = 397
set osd_link_quality_pos = 2735
set osd_rssi_dbm_pos = 2743
set osd_tim_2_pos = 435
set osd_throttle_pos = 456
set osd_mah_drawn_pos = 467
set osd_craft_name_pos = 2076
set osd_warnings_pos = 14746
set osd_avg_cell_voltage_pos = 2749
set osd_displayport_device = MSP
set vcd_video_system = HD
set gyro_1_align_yaw = 1800
set craft_name = APEX5
set pilot_name = CEDRIC

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 82
set dterm_lpf1_dyn_max_hz = 165
set dterm_lpf1_static_hz = 82
set dterm_lpf2_static_hz = 165
set p_pitch = 56
set i_pitch = 70
set d_pitch = 32
set f_pitch = 149
set p_roll = 53
set i_roll = 67
set d_roll = 28
set f_roll = 143
set p_yaw = 53
set i_yaw = 67
set f_yaw = 143
set d_min_roll = 28
set d_min_pitch = 32
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 45
set feedforward_jitter_factor = 10
set feedforward_max_rate_limit = 95
set dyn_idle_min_rpm = 30
set simplified_master_multiplier = 120
set simplified_i_gain = 70
set simplified_d_gain = 80
set simplified_dmax_gain = 0
set simplified_dterm_filter_multiplier = 110

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 85
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 65
set pitch_srate = 65

rateprofile 1

# rateprofile 1
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_srate = 20
set pitch_srate = 20
set yaw_srate = 20

rateprofile 2

# rateprofile 2
set roll_rc_rate = 9
set pitch_rc_rate = 9
set yaw_rc_rate = 9
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 70
set pitch_srate = 68
set yaw_srate = 50

rateprofile 3

# restore original rateprofile selection
rateprofile 2

# save configuration
```

Canaill51, 2022-12-23:
I finally understand what I am doing wrong. When I flashed the FC, I did not select "Classic Mode".
I just reflashed it with "Classic Mode" enabled, then loaded my backup and no more error messages as above.

haslinghuis, 2022-12-23:
You did nothing wrong. Configuration has some definitions for `BARO`, `MAG` and `PINIO`

Please add these options to `Other Options` before flashing. If you don't need these options you can just ignore the errors.

Canaill51, 2022-12-23:
I just tried your suggestion:
[image]
[image]

Here, I have another error: PPM.
But anyway, as you said, I don't need these options.

I think "Classic Mode" is easier for basis --> This would be set as default on BF GUI Release?

haslinghuis, 2022-12-23:
Would be solved with `PPM` in other options :)

We like to solve issues with cloud build as it will be mandatory for 512K board soon as flash is limited for further development

McGiverGim, 2022-12-23:
I think is a good idea. Classic as default if the space is enough for this release. We are complicating things when is not necessary.

aowi7280, 2023-01-10:
You could also show what is included in the Classic mode. Then a user can remove what he knows he doesn't need and leave anything he is unsure of.

haslinghuis, 2023-01-29:
This problem does not relate to configurator as we are using CLI it's an firmware issue. Most of the errors has been filtered out by now

closed 2023-01-29 by haslinghuis