betaflight/betaflight-configurator #3162: Target tooltip documentation requires updating.

hydra, 2022-12-25:
### Describe the bug

Given the new cloud-build system this appears a little out of date now:

[image]

### To Reproduce

Hover over the tooltip.

### Expected behavior

Up-to-date documentation.

### Configurator version

`master`

haslinghuis, 2022-12-29:
Made a new wiki page here: https://github.com/betaflight/betaflight/wiki/Flashing-firmware

Appreciate any input on the tooltip here:

```
    This version of Betaflight configurator supports flashing of Unified Targets with the respective board specific configurations in one step.
    The different firmware types that are available for each board are shown in the drop-down as follows:<br /><br /><strong>&lt;board name&gt;</strong>

    <strong>&lt;board name&gt; (&lt;manufacturer id&gt;)</strong>:<br>(4 character manufacturer id)<br>Unified Target.<br /><br /><strong>
     
    Starting with version 4.4, Betaflight introduces <strong>Cloud Build</strong>
    
    With Cloud Build we need to define hardware options present on your build.
    To make the different boards work with the same firmware, a specific configuration file is deployed alongside the firmware when a Unified Target is flashed.<br>
    
    With Local build you can load a unified target configuration file or chose a board before loading a firmware .hex file.

```

HThuren, 2022-12-29:
My suggestion:
----
    This version of Betaflight configurator supports flashing of Unified Targets with the respective board specific configurations in one step.
    The concept of Unified Targets means that the same firmware .hex file can be used for all boards using the same MCU.

    The different firmware types that are available for each board are shown in the drop-down as follows:<br /><br /><strong>&lt;board name&gt;</strong>
    <strong>&lt;board name&gt; (&lt;manufacturer id&gt;)</strong>:<br>(4 character manufacturer id)<br>Unified Target.<br /><br /><strong>
     
    Starting with version 4.4, Betaflight introduces <strong>Cloud Build</strong>

    With Cloud Build we need to define hardware options present on your build.
    To make the different boards work with the same firmware, a specific configuration file is deployed alongside the firmware when a Unified Target is flashed.<br>
    
    With Local build you can load a unified target configuration file or chose a board before loading a firmware .hex file.

    If you encounter problems using the firmware, please open an <a href="https://github.com/betaflight/betaflight/issues" target="_blank" rel="noopener noreferrer">issue</a>.

closed 2022-12-30 by haslinghuis

hydra, 2023-01-09:
Just a reminder that the change in #3175 will have to be reviewed again when "the functionality of loading a config file will be removed from the load local button", see: https://github.com/betaflight/betaflight-configurator/issues/3172#issuecomment-1365910776

PR #3175 (2022-12-30): Fix tooltip
Fixes: #3162 
Fixes: #3169