betaflight/betaflight-configurator #3171: Odd behaviour when toggling 'Release' and 'Release and Release Candidates'

hydra, 2022-12-27:
### Describe the bug

UI elements shown incorrectly.

### To Reproduce

When you first select 'Release and Release Candidate' the UI looks like this:

[image]

If you then toggle classic mode it looks like this:

[image]
a

If you then select 'Release' it looks like this:

[image]

When you select 'Release and Release Candidate' again the UI looks like this:

[image]

### Expected behavior

Consistent UI elements based on selections.

### Configurator version

10.9.0-debug-23b74d99

blckmn, 2022-12-27:
This is currently because there is no cloud build release that is classified as a release.

Once 4.4.0 is released then the UI will function as expected.

For clarity, 4.3.2 or lower is not cloud built, but rather uses the pre built targets. Therefore there's no need to show the option panel as it serves no purpose.

hydra, 2023-01-29:
@blckmn This seems to be fixed now.  I just updated `master` and I'm unable to reproduce this.

closed 2023-01-29 by hydra