betaflight/betaflight-configurator #3172: Unable to access Build Configuration or flash a target using a '.config' file.

hydra, 2022-12-27:
### Describe the bug

The UI does not show any cloud build options (Build Configuration UI block) when given a local `<my target>.config` which includes the base target name, nor when the 'autodetect' correctly identifies the base target if one has been built locally and flashed.

### To Reproduce

1. Prepare a 'unified target config' file, e.g. 'SPRO-SPRACINGH7EXTREME.config' as per unified target instructions, which also includes `#define` statements which necessitate a firmware build by the cloud build system.
2. Go to the 'Firmware Flasher' page.
3. Click 'Load firmware local', select a the `.config` file.

[image]

4. Observe the progress bar shows 'Loaded target, please load firmware file'.

[image]

5. Clicking 'Load firmware' causes 'No selected firmware to load' to appear in the log.

[image]

6. Clicking 'Auto detect' will show this, as the target hasn't been published yet:

[image]

Observe that the base target is now also known to and shown by the configurator (STM32H750).

### Expected behavior

a) At step 4. above, it would be helpful if the base target (STM32H750 in this case) was shown in the 'Choose a Board' drop-down, the `.config` file contains this information:

```
# Betaflight / STM32H750 (SP7E) 4.4.0 Dec 26 2022 / 12:28:11 (7671c7eb01) MSP API: 1.45
```

It would then be helpful if the Build Configuration UI block was also shown so that you can select a commit to build firmware from.

b) Similarly, at Step 6, which occurs if the user has built the firmware locally and flashed it, it would be again helpful to select the base target (STM32H750) in the same place in the UI and again show the Build Configuration UI block.

### Configurator version

10.9.0-debug-dc2880fc

### Add any other context about the problem that you think might be relevant here

It's also not clear what the configurator will do with the `.config` file that was loaded if other UI buttons are pressed or if a `.hex` file is subsequently picked via 'Load Firmware [Local]'.

hydra, 2022-12-27:
Ping @blckmn

blckmn, 2022-12-27:
The functionality of loading a config file will be removed from the load local button.

A UI element like that shouldn't be dual purpose.

The local config loading will have its own method. Yet to be determined.

There is currently no ability to Load a local config with #defines. Conceivably we could enable expert mode if they're detected and add them to the custom defines options.

If loading a local hex you'd have to select the correct target to gets it's config from online. There's no value in having targets that are merely the MCU as they'd have no configuration anyway. If loading a local target and config then target selection is moot, and conceivably be hidden.

This whole page needs a redesign as the UI is not right. I mean there are flash connectivity options miles from the flash button.

My view I'd like everything in panels, and you go through step by step, nothing is shown until you've progressed to that point.

hydra, 2022-12-27:
> The functionality of loading a config file will be removed from the load local button.

OK.

> A UI element like that shouldn't be dual purpose.

I agree.

> The local config loading will have its own method. Yet to be determined.

OK.

> There is currently no ability to Load a local config with #defines. Conceivably we could enable expert mode if they're detected and add them to the custom defines options.

Yes, but using the existing somewhat hacky approach of a single .txt file with dual-purpose (defines and cli commands) and accidentally similar comment structure (# = comment in cli, but #define means something to cloud build parser) is probably not a good idea.  suggest converting all the unified target configs to json files and using an appropriate json structure or at least removing the #defines from the .config file into a .json file along with other settings, such as the base-target that it applies too, linker settings, and other command-line arguments, etc.  e.g. `SPRO-SPRACINGH7EXTREME.json` only, or `SPRO-SPRACINGH7EXTREME.json` coupled with `SPRO-SPRACINGH7EXTREME.config`.

> If loading a local hex you'd have to select the correct target to gets it's config from online. There's no value in having targets that are merely the MCU as they'd have no configuration anyway. If loading a local target and config then target selection is moot, and conceivably be hidden.

Sort of, I agree that if none of the MCU base targets are ever meant to be booted, including the H743, H7x2, etc then by all means keep them out of the list of targets but somehow the build system (e.g. from a .json file) needs to be able to be told what base target to use.  Currently though the /some/ of the MCU base targets ARE meant for booted.  So we need to go fully one way or the other to set both developer and user expectations.
 
> This whole page needs a redesign as the UI is not right. I mean there are flash connectivity options miles from the flash button.

agreed.
 
> My view I'd like everything in panels, and you go through step by step, nothing is shown until you've progressed to that point.

agreed.

hydra, 2023-01-29:
ping @blckmn please see https://github.com/betaflight/betaflight-configurator/issues/3172#issuecomment-1366014662

closed 2023-03-09 by github-actions[bot]