betaflight/betaflight-configurator #3173: Release notes / build information not shown in the UI.

hydra, 2022-12-27:
### Describe the bug

The build information panel no longer appears when selecting any firmware.

### To Reproduce

Select firmware, observe missing panel:

[image]

### Expected behavior

Panel showing build information, filename, date, release notes, etc, should be shown.

### Configurator version

10.9.0-debug-dc2880fc

blckmn, 2022-12-27:
This is an intentional omission.

hydra, 2022-12-27:
Is there a PR somewhere that discussed the reasons for removing it?  I and probably others would like to know what happened to it and why.

haslinghuis, 2022-12-27:
See https://github.com/betaflight/betaflight-configurator/pull/3083

closed 2022-12-27 by haslinghuis