betaflight/betaflight-configurator #3186: No clue what im doing please help

Shiggle, 2022-12-30:
### Describe the bug

im trying to get betaflight on my acer chromebook and i downloaded the debian version for my linux build (i know nothing about linux i struggled to figure out how to find the right version) I keep getting an error when i hit install [Error while installing package: dpkg-deb --control subprocess returned error exit status 2] i used the chromebook version of betaflight and now my drone wont work.. no clue what i did 

### To Reproduce

i have no clue as i said

### Expected behavior

na

### Configurator version

na

### Flight controller configuration

```txt
na
```

### Add any other context about the problem that you think might be relevant here

na

nerdCopter, 2023-01-16:
I don't have chromebook-specific experience, but only newish chromebooks can run Linux packages.
please see the following resource. If it does not work for you, you may have to find alternate solutions. (i.e. BF's Android .APK or 3rd-party SpeedyBee App)

https://www.fossmint.com/install-linux-apps-on-chromebook/

Shiggle, 2023-01-16:
still having trouble now. Ive got the app working now ! Buuuut the app
doesnt even read the drone connected with and without batteries plugged in

Thanks for the previous help!

nerdCopter, 2023-01-17:
USB cable must provide power to FC (usually always the case on standard PC's).
Try other branded/model USB cables. Otherwise, Sometimes 5v regulator on FC may be damaged. preventing power-by-USB.

Shiggle, 2023-01-18:
I tried multiple new cables and the drone lights up and that's it. App
still doesn't read the drone. does that mean I mean I need a new FC?

McGiverGim, 2023-01-18:
@Shiggle GitHub is for development issues. If you have user problems, please go to our official chat where some people will help you without problem https://discord.gg/n4E6ak4u3c

closed 2023-01-18 by nerdCopter