betaflight/betaflight-configurator #3202: HD OSD  data not  extending to right black region

zachninan, 2023-01-03:
### Describe the bug

Noticed the OSD data while in HD mode seems not to extend beyond the video region. 
It is expected to move to the black region on the right side . The left side OSD data does move to the black vacant region .
 
Fix : 
Now when we click on the preview pane a grid appears . This expends outside the OSD preview pane . Now we can move the right data which goes outside the preview pane . Now if we view in the goggles the data will appear in the right  black  region . 

Hardware : 
This was tested on the V2 goggles and caddx vista running WTFOS 0.9 . 
BF Configurator version :  10.9.0 RC3 

### To Reproduce

Move OSD data to the right top / bottom corner  with betaflight OSD tab within the preview pane . 
Connect and view in the goggles . 

### Expected behavior

Data expected to move to the right black region within the goggles . 

### Configurator version

10.9.0 RC3

### Flight controller configuration

```txt
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.30V, Core temp=31degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3604, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MSP (60 x 22)
BUILD KEY: 4.4.0-RC3/GEPRCF411/804a5c96e1553630a2a2aa58f5a424b1
System Uptime: 68 seconds, Current Time: 2023-01-03T06:15:14.407+00:00
CPU:45%, cycle time: 314, GYRO rate: 3184, RX rate: 15, System rate: 9
Voltage: 413 * 0.01V (1S battery - OK)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP
# get osd
osd_units = METRIC
Allowed values: IMPERIAL, METRIC, BRITISH

osd_warn_bitmask = 8191
Allowed range: 0 - 4294967295

osd_rssi_alarm = 20
Allowed range: 0 - 100

osd_link_quality_alarm = 80
Allowed range: 0 - 100

osd_rssi_dbm_alarm = -60
Allowed range: -130 - 0

osd_rsnr_alarm = 4
Allowed range: -30 - 20

osd_cap_alarm = 2200
Allowed range: 0 - 20000

osd_alt_alarm = 100
Allowed range: 0 - 10000

osd_distance_alarm = 0
Allowed range: 0 - 65535

osd_esc_temp_alarm = 0
Allowed range: 0 - 255

osd_esc_rpm_alarm = -1
Allowed range: -1 - 32767

osd_esc_current_alarm = -1
Allowed range: -1 - 32767

osd_core_temp_alarm = 70
Allowed range: 0 - 255

osd_ah_max_pit = 20
Allowed range: 0 - 90

osd_ah_max_rol = 40
Allowed range: 0 - 90

osd_ah_invert = OFF
Allowed values: OFF, ON

osd_logo_on_arming = OFF
Allowed values: OFF, ON, FIRST_ARMING

osd_logo_on_arming_duration = 5
Allowed range: 5 - 50

osd_tim1 = 2560
Allowed range: 0 - 32767

osd_tim2 = 2561
Allowed range: 0 - 32767

osd_vbat_pos = 2497
Allowed range: 0 - 65535
Default value: 234

osd_rssi_pos = 234
Allowed range: 0 - 65535

osd_link_quality_pos = 3606
Allowed range: 0 - 65535
Default value: 234

osd_link_tx_power_pos = 234
Allowed range: 0 - 65535

osd_rssi_dbm_pos = 234
Allowed range: 0 - 65535

osd_rsnr_pos = 234
Allowed range: 0 - 65535

osd_tim_1_pos = 234
Allowed range: 0 - 65535

osd_tim_2_pos = 3574
Allowed range: 0 - 65535
Default value: 234

osd_remaining_time_estimate_pos = 234
Allowed range: 0 - 65535

osd_flymode_pos = 2433
Allowed range: 0 - 65535
Default value: 234

osd_anti_gravity_pos = 234
Allowed range: 0 - 65535

osd_g_force_pos = 234
Allowed range: 0 - 65535
```

McGiverGim, 2023-01-03:
Attaching since configurator screen always help. I think I have seen it at discord?

zachninan, 2023-01-03:
> Attaching some configurator screen always help. I think I have seen it at discord?
[image]

zachninan, 2023-01-03:
[image]

McGiverGim, 2023-01-03:
Thanks! What is your screen resolution?

zachninan, 2023-01-03:
> Thanks! What is your screen resolution?

1920 x 1080

zachninan, 2023-01-03:
Is this a BF standalone configurator  for windows ?  If there is one such ,
i can test  and provide feedback .

Dont think i will be able to build a binary or experiment much due to a
full day job though .

McGiverGim, 2023-01-03:
Yes, the builds for new PRs contains the standalone version too.

zachninan, 2023-01-03:
I used the standalone version and tested right away . Looks like  now the grid does not go outside the preview pane. Seems to work right now . 
Additionally i wanted to mention that while enabling OSD element such as the cross hair , Artificial horizon etc , it does not center them on the preview pane . It goes towards the left side . Please check this one also if it has a relation . 
Appreciate the quick fix on this one .

closed 2023-01-04 by haslinghuis

GootyS, 2023-05-29:
I have this issue as well, osd fits nice to the left but not on the right. In BF configurator 10.10.0 nightly build the OSD shows its fits as i want it to but in the goggles its all the way left but not on the right

commit 20e3abc7 (2023-01-04): Fix for OSD grid overflowing the preview area as per #3202 (#3203)

Fix for OSD grid overflowing the preview error due to the invdividual images not shrinking below a fixed width. Fixes #3202 / https://github.com/betaflight/betaflight-configurator/issues/3202